LibreCAD/LibreCAD #1637: Compilation error in QG_GraphicView

gerardbm, 2023-06-20:
### Expected behavior

### Observed behavior

Error log:

```
ui/qg_graphicview.cpp: In member function ‘virtual void QG_GraphicView::tabletEvent(QTabletEvent*)’:
ui/qg_graphicview.cpp:507:20: error: ‘class QTabletEvent’ has no member named ‘deviceType’; did you mean ‘device’?
         switch (e->deviceType()) {
                    ^~~~~~~~~~
ui/qg_graphicview.cpp: In member function ‘virtual void QG_GraphicView::wheelEvent(QWheelEvent*)’:
ui/qg_graphicview.cpp:609:34: error: ‘class QWheelEvent’ has no member named ‘position’; did you mean ‘posted’?
     RS_Vector mouse = toGraph(e->position().x(), e->position().y());
                                  ^~~~~~~~
ui/qg_graphicview.cpp:609:53: error: ‘class QWheelEvent’ has no member named ‘position’; did you mean ‘posted’?
     RS_Vector mouse = toGraph(e->position().x(), e->position().y());
                                                     ^~~~~~~~
ui/qg_graphicview.cpp:794:45: error: ‘class QWheelEvent’ has no member named ‘position’; did you mean ‘posted’?
                                          e->position(),
                                             ^~~~~~~~
make[2]: *** [../../generated/librecad/obj/qg_graphicview.o] Error 1
make[1]: *** [sub-src-make_first] Error 2
make: *** [sub-librecad-make_first-ordered] Error 2
```

This error comes from some commits ago. I think it's easy to fix.

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

Debian 9.

dxli, 2023-06-21:
this one should have been fixed in master.

gerardbm, 2023-06-21:
The error changed to another one:

```ui/qg_graphicview.cpp: In member function ‘virtual void QG_GraphicView::wheelEvent(QWheelEvent*)’:
ui/qg_graphicview.cpp:782:5: error: no matching function for call to ‘QMouseEvent::QMouseEvent(<brace-enclosed initializer list>)’
     };
     ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtGui/QNativeGestureEvent:1:0,
                 from ui/qg_graphicview.cpp:34:
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:114:5: note: candidate: QMouseEvent::QMouseEvent(QEvent::Type, const QPointF&, const QPointF&, const QPointF&, Qt::MouseButton, Qt::MouseButtons, Qt::KeyboardModifiers, Qt::MouseEventSource)
     QMouseEvent(Type type, const QPointF &localPos, const QPointF &windowPos, const QPointF &screenPos,
     ^~~~~~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:114:5: note:   candidate expects 8 arguments, 5 provided
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:111:5: note: candidate: QMouseEvent::QMouseEvent(QEvent::Type, const QPointF&, const QPointF&, const QPointF&, Qt::MouseButton, Qt::MouseButtons, Qt::KeyboardModifiers)
     QMouseEvent(Type type, const QPointF &localPos, const QPointF &windowPos, const QPointF &screenPos,
     ^~~~~~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:111:5: note:   candidate expects 7 arguments, 5 provided
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:108:5: note: candidate: QMouseEvent::QMouseEvent(QEvent::Type, const QPointF&, const QPointF&, Qt::MouseButton, Qt::MouseButtons, Qt::KeyboardModifiers)
     QMouseEvent(Type type, const QPointF &localPos, const QPointF &screenPos,
     ^~~~~~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:108:5: note:   candidate expects 6 arguments, 5 provided
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:106:5: note: candidate: QMouseEvent::QMouseEvent(QEvent::Type, const QPointF&, Qt::MouseButton, Qt::MouseButtons, Qt::KeyboardModifiers)
     QMouseEvent(Type type, const QPointF &localPos, Qt::MouseButton button,
     ^~~~~~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:106:5: note:   no known conversion for argument 2 from ‘<brace-enclosed initializer list>’ to ‘const QPointF&’
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:103:20: note: candidate: QMouseEvent::QMouseEvent(const QMouseEvent&)
 class Q_GUI_EXPORT QMouseEvent : public QInputEvent
                    ^~~~~~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:103:20: note:   candidate expects 1 argument, 5 provided
make[2]: *** [../../generated/librecad/obj/qg_graphicview.o] Error 1
make[1]: *** [sub-src-make_first] Error 2
make: *** [sub-librecad-make_first-ordered] Error 2
```

I don't know if it's related or if it's a different thing.

lordofbikes, 2023-06-21:
This is probably because of the Qt version.
Debian 9's qt5-default is version 5.11.3.
I'm not happy with the new requirement for Qt 5.15.
This makes building from source more complicated or impossible on older systems.
And I know, that there is a broad user base, using older systems for their LibreCAD installation in workshops.
E.g. we have still more than 300 downloads/month for the Windows XP installer, which is bound to Qt 5.6. This is of course not a majority, but it shows, that there are users out there who are happy to still have it.

So I advocate to revert that barrier, to stay with a lesser Qt version, because the benefits are mostly in code cosmetics and not needed for performance, stability or usability.
Or at least implement it in an optional way, to keep the possibility to build LibreCAD on older systems.

If I'm alone with this point of view, it needs at least a guard in the project file, which stops `qmake` immediately with Qt versions below Qt 5.15, to avoid irritations like this issue.
The consequence is, that users of older systems have to stay with last stable release 2.2.0.

dongxuli-concord-design, 2023-06-21:
We can branch out a version for old qt5, or if the change is small enough, in the same branch with some version based code.

5.15-lts is the oldest release still being supported, so the choice is not really there to keep the support for older versions.

lordofbikes, 2023-06-22:
My point is not what's supported or not. Debian 9 is EOL too, but as we can see, it is still in use.
LibreCAD is used by makers and hobbyists, and as such one myself, I know that it is a hard decision to touch a running system, just because it is declared deprecated.

It makes a difference, if a mandatory version is declared, because critical functions have been removed or because of the use of new classes or language features which are not available in previous versions. It's probably not only Qt versions, but also the contained tool chain, namely the compiler versions which are not capable to build LibreCAD anymore.

What I see is the second case for LibreCAD code base now and it's just to avoid using them in a few places to keep backward compatibility.

I don't see the resources for an extra pre Qt 5.15 branch. Either we manage to keep the `master` branch working for pre Qt 5.15 environments or users of these systems have to use CI binaries in future and have to stop building from code themselves.
In this case, as already said, `qmake` should already fail with an error about the deprecated Qt version. When the `make` process fails many users can't interpret the compiler errors which are not obvious related with Qt version.

gerardbm, 2023-06-22:
Interesting points. I like the idea of having the ability to compile LibreCAD. At the same time, I like the innovation. So I have some questions about it.

The use of Qt 5.15: the binary EXE for Windows XP will work?

The use of Qt 5.15: an AppImage will work in older systems?

What does the version Qt 5.15 over the version 5.6?

Is it difficult to do a regression in an older Qt version?

What would be lost in this regression? What are we gaining with the version Qt 5.15?

Having the AppImage is really a good improvement, since it can run in a lot of systems without compiling and without installation. This is avoiding the need to compile LibreCAD everytime.

dongxuli-concord-design, 2023-06-22:
WinXP might be too old for the windows 32 bit CI build, but I have not tested it.

For Linux and Mac, the latest CI builds are expected to work for a wider range of versions. We may want to collect test results and provide a list of not supported cases.

What I want to avoid is the case it would only build on old unsupported versions( Qt and Os). When it fails to build on the oldest long term support versions, it appears to be an abandoned upstream to package maintainers.

We can support old OSs with old releases, and the current OSs with the latest development. To mix the old OSs with the latest development should rely on manual backporting. To support end of live OSs with the latest development is good, but shouldn't be a priority over support for the current OSs.

Back to the original question, what's the gain of the code change. The goal is to make the code build with an oldest supported Qt release.

gerardbm, 2023-06-23:
> WinXP might be too old for the windows 32 bit CI build, but I have not tested it.

I tested it with the binary `LibreCAD-Installer-2.2.1_alpha-188-g5c4d14de.exe`. The installation was successful, but the program is not working.

> What I want to avoid is the case it would only build on old unsupported versions( Qt and Os). When it fails to build on the oldest long term support versions, it appears to be an abandoned upstream to package maintainers.

Totally true. LibreCAD needs to be updated enough to be an actractive software.

I would give up compiling LibreCAD if it is improved instead. I love every commit you do on it, because now LibreCAD is getting better every day.

closed 2023-07-08 by dxli