LibreCAD/LibreCAD #1638: Text rendering issue with French accent

fa201, 2023-06-21:
### Expected behavior
French accent is always properly rendered.

### Observed behavior
A DXF file generated with one of 2.2.0 release candidates has some French accent `é` not rendered correctly at file opening. `'` is rendered with the `e` below.

### Steps to reproduce or sample file
Open the file.
Show all layers (layers with text were hidden and locked when the file was saved). See picture.
Hide the text layer (still locked)
Show the text layer: now the text is OK. See picture.

### Operating System and LibreCAD version info
Version: 2.2.1_alpha-188-g5c4d14de
Compiler: GNU GCC 11.3.0
Compiled on: Jun 20 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.2 LTS
CI AppImage
[image]
[image]

Notes:
- switching on/off another layer without text give OK result.
- redraw does not help

dxli, 2023-06-21:
This appears to be a hard one. Conceivable workaround could be larger text sizes, higher dpi at rendering, or another font

fa201, 2023-06-22:
The work around is to show on/off some layers which is faster without any impact on the drawing.

dxli, 2023-06-23:
Can you upload a test case dxf file?

fa201, 2023-06-24:
Unfortunately, I am not allowed to share this file.

closed 2023-06-27 by dxli