LibreCAD/LibreCAD #1639: Target shape appear on last AppImage

fa201, 2023-06-21:
### Expected behavior
There is no target shape as shown on picture generated by 
Version: 2.2.0
Compiler: GNU GCC 9.4.0
Compiled on: Dec 17 2022
Qt Version: 5.12.9
Boost Version: 1.75.0
System: Ubuntu 22.04.2 LTS
CI AppImage
[image]

### Observed behavior
A target shape appear with a constant size (in pixels not in mm). See pictures with increased zoom for each picture.
Version: 2.2.1_alpha-188-g5c4d14de
Compiler: GNU GCC 11.3.0
Compiled on: Jun 20 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Ubuntu 22.04.2 LTS
CI AppImage
[image]
[image]
[image]
![shape_NOK4](https://github.com/LibreCAD/LibreCA
[image]
D/assets/49532187/bf2f565b-1969-43ca-8857-90392beb6233)

Note that this shape can be selected as shown on the last picture.

dxli, 2023-06-21:
This is a new feature: customized point shapes

Please have a look at drawing preferences, and there's a new tab "Points", and the point outlines can be selected there.

I will review the default choice in code.

dxli, 2023-06-24:
The current default point outline and size are defined in lc_defaults.h

```C++

/**
 * Defines default values for system settings etc within the
 * LibreCAD program. 
 * Eg. Application preferences default values, current drawing preferences
 * default values, 
 *
 * @author Polly Winters
 */

/*	Point display mode and size  */
#define LC_DEFAULTS_PDMode		DXF_FORMAT_PDMode_EncloseCircleSquare(DXF_FORMAT_PDMode_CentreDot)
#define LC_DEFAULTS_PDSize		        DXF_FORMAT_PDSize_Percent(2.0)
```

Since both the default shape and size could be set by the end user in:

Edit > Current drawing preferences > Points > Point Size

it's not considered a regression.

closed 2023-06-24 by dxli