LibreCAD/LibreCAD #1640: With small scale, the selection mouse over is huge

yageek, 2023-06-23:
I wanted to simply draw lines and hatches.

Every time the cursor is over one element, a huge red overlay is displayed.

Did I misconfigured something here?

### Steps to reproduce or sample file
[terrasse.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/11847525/terrasse.dxf.zip)

### Operating System and LibreCAD version info
macOS 14.4 LibreCad 2.2.1_alpha-188-g5c4d14de
[image]

dxli, 2023-06-23:
This is due to the new glowing effect for mouse hovering.

The code quality is still not robust enough:

1. it fails to separate the visual effect layer from the actual geometry layer;
2. it may cause performance issues by creating thousands of entities for the visual effect of glowing;
3. it computes visual effect sizes in the geometry units, causing scaling issue(like this particular issue).

dxli, 2023-06-23:
A fix has been merged.

dxli, 2023-06-23:
TODO: still need to separate the visual effects from the core geometry layer. The current design of the glowing effect is too dangerous for data integrity.

closed 2023-06-24 by dxli

dellus, 2023-06-29:
Are you sure this is really fixed? See Issue #1645.