LibreCAD/LibreCAD #1645: Strange highlighting elements on hover

PeterHagen, 2023-06-29:
### Expected behavior

Not sure what the expected behaviour should be, but I guess the element where the mouse is hovering about is highlighted

### Observed behavior

See image: 

[image]

### Steps to reproduce or sample file

Any file will reproduce

### Operating System and LibreCAD version info

macOS (M1) 13.4.1

LibreCAD
Version: 2.2.1_alpha-212-g9ad76aa2
Compiler: Clang 14.0.3
Compiled on: Jun 29 2023
Qt Version: 5.15.10
Boost Version: 1.76.0
System: macOS 13.4

dellus, 2023-06-29:
I have discovered the size of the effect depends on the line width. It is OK with line width 0.00 mm and grows with larger, real widths.

dxli, 2023-06-30:
@dellus confirmed.

So, we have two issues:

1, width control: currently the width could be too much. The goal, the glowing effect shouldn't be larger than double the displayed width;
2, The "Selected" and "Highlighted" colors messed up: "Selected"(glowing) color is displayed by the "Highlighted" color.

dxli, 2023-06-30:
Included in 951c845

1, Glowing effects are using the color of selected unitities, instead of highlighted, which is reserved for actions, such as trimming;
2, Glowing effects screen width is limited to certain maximum, which may need to be further optimized.

PeterHagen, 2023-07-07:
I'm personally not happy with the highlighting. This is a screenshot of the latest master:

[image]

As far as I know I didn't make any user settings for the highlighting, so this should be with default values. In this case it's just one text block, but in case of a block, everything turns into a grey blur.

Previously, there was no highlighting as far as I can recall. Can this feature be made optional?

dongxuli-concord-design, 2023-07-07:
Yes, it shouldn't be disruptive.

I will add an option to enable it.

fa201, 2023-07-07:
@dxli I agree with @PeterHagen . I was about to open an issue to ask for a check box for this feature as I find it annoying for some works.

PeterHagen, 2023-07-08:
Perfect! Thanks for the fix!

closed 2023-07-08 by dxli