LibreCAD/LibreCAD #1648: Some hatch tyles produces not seamless pattern

app4soft, 2023-07-07:
### Expected behavior

Hatch tiles should produce seamless pattern.

### Observed behavior

Some hatch tiles produces not seamless pattern. For example, built-in `gost_metal`:

- https://github.com/LibreCAD/LibreCAD/blob/master/librecad/support/patterns/gost_metal.dxf

[image]

### Steps to reproduce or sample file
1. Create any shape wider than hatch tile size;
2. Fill it with "Hatch" tool using pattern `gost_metal`;
3. Check tile edges stitching.

### Operating System and LibreCAD version info

- Version: `2.2.1_alpha-214-`ga6f34238 (AppImage)

app4soft, 2023-07-10:
To clarify issue, _LibreCAD_ uses `100 mm × 100 mm` tile size, but some patterns can't be seamless in such size.

For `gost_metal.dxf` (`45°` lines with `5 mm` distance) tile should be cropped (not scaled) to `99 mm × 99 mm` to produce seamless pattern.

So, LibreCAD should has an option to support other tile size too.

Or is there some limitation on tile size in DXF spec?

app4soft, 2023-07-10:
After some more study, I've found that LibreCAD ignores pattern tile origin, and fits tile to visible entities pattern from all sides — but it should fit from only top and bottom. That might be a bug.

- `hex.dxf`
  [image]

- `clay.dxf`
  [image]

dxli, 2023-07-10:
That means the bounding box of a tile is always recalculated, instead of a hard coded 100mmx100mm.

Of course, for each 100x100 tile, geometries should b drawn treating the tile as a flat torus

app4soft, 2023-07-10:
> That means the bounding box of a tile is always recalculated, instead of a hard coded 100mmx100mm.

Hardcoding tile size to `100mm×100mm` is bad idea at all, because, as said above, some patterns may require other tile dimension (and not only in size, but also in a form of non-square rectangle).

But there are two issue:

1. Actual issue is about hatch patterns tiles content design;
2. Another issue (see https://github.com/LibreCAD/LibreCAD/issues/1652) is about how LibreCAD handle hatch pattern tile files.

luzpaz, 2023-10-11:
typo in title, should be 'tiles' not 'tyles'

u2fly, 2025-04-27:
> That means the bounding box of a tile is always recalculated, instead of a hard coded 100mmx100mm.

Keep it as is for now, as it allows non-square tiles with custom size.

Need to correct existing patterns' itself.

[image]

[image]

closed 2025-04-27 by u2fly