LibreCAD/LibreCAD #1653: Crashing when I click in the drawing area

dvanw6, 2023-07-12:
### Expected behavior
Not crashing when I click in the drawing area
### Observed behavior
Crashing when I click in the drawing area
### Steps to reproduce or sample file
Opening a 541kb .dwg file exported from Autocad LT 2008, then clicking anywhere in the drawing area. (Cannot provide file, NDA work)
### Operating System and LibreCAD version info
Windows 10
Ryzen 5 5500
32 GB RAM
GTX 1660
LibreCAD 2.2.0.1

dvanw6, 2023-07-12:
note, exported a .dxf version and that also crashed

lordofbikes, 2023-07-12:
Because the DWG/DXF file in question is essential for this issue, it can't be further analyzed without it.
Maybe you can modify the drawing for a test case file? E.g. by removing layers or parts of the drawing. The issue must persist and trigger the issue repeatable, then we can investigate the cause.

Feel free to reopen this issue, when you can provide a file to reproduce the crash behavior.

dvanw6, 2023-07-12:
> Because the DWG/DXF file in question is essential for this issue, it can't be further analyzed without it. Maybe you can modify the drawing for a test case file? E.g. by removing layers or parts of [...]

Am I able to directly send you the file?

lordofbikes, 2023-07-12:
Sure, send it to `armin(at)librecad.org` and it will hit only my mailbox.
Send both files if possible, DWG and DXF.

dvanw6, 2023-07-12:
Sent

lordofbikes, 2023-07-12:
A reduced test case to trigger the issue:
[TestCase.dxf.txt](https://github.com/LibreCAD/LibreCAD/files/12029253/TestCase.dxf.txt)

Open the test case file and left click around X-670 Y700, this triggers the segfault causing the crash.

https://github.com/LibreCAD/LibreCAD/blob/209d4690cd07c7ef200892603892bdb279d7f5fd/librecad/src/lib/math/rs_math.cpp#L599-L604
In line 599, the return value of `quadraticSolver()` is empty, what results in a segfault in line 604, where `r[0]` is accessed.
So the question is, why the result is empty, what `//should not happen`.

closed 2023-07-16 by dxli