LibreCAD/LibreCAD #1654: MText object shows only first line of multiline text

app4soft, 2023-07-13:
### Expected behavior

MText object should show all multi-line text as it inputed.

### Observed behavior

MText object shows only first line of multi-line text input.

In same time when move MText object bounding box shows as it include text of multi-line text input.

[image]

### Steps to reproduce or sample file

1. Create "MText" object;
2. Input multi-line text:
  ```
  R/L*360°
  25/74.33*360°
  121.08°
  ```
3. Press <kbd>OK</kbd>

### Operating System and LibreCAD version info

- **Version:** `2.2.1_alpha-222-gfc51494d` (GitHub's AppImage)
- **Compiler:** GNU GCC 11.3.0
- **Compiled on:** Jul 13 2023
- **Qt Version:** 5.15.3
- **Boost Version:** 1.74.0
- **System:** Debian GNU/Linux 12 (bookworm)

efraletti, 2023-07-13:
Hi, I'm having this exact issue, only first line shows, even in "print preview"

I'm using

Version: 2.2.1_alpha-224-gc8389f010
Compiler: GNU GCC 13.1.1
Compiled on: Jul 13 2023
Qt Version: 5.15.10
Boost Version: 1.81.0
System: Arch Linux

dongxuli-concord-design, 2023-07-13:
I will look into this one.
Just in case, could you test whether this is a regression, say from 2.1.3

app4soft, 2023-07-13:
> Just in case, could you test whether this is a regression, say from 2.1.3

May You provide AppImage for `2.1.3`? (and add it to related release assets)

efraletti, 2023-07-14:
last commit fixed it! thank you!

```
Version: 2.2.1_alpha-231-g3974b2205
Compiler: GNU GCC 13.1.1
Compiled on: Jul 14 2023
Qt Version: 5.15.10
Boost Version: 1.81.0
System: Arch Linux
```

app4soft, 2023-07-14:
Confirming: issue fixed now.

[image]

closed 2023-07-14 by app4soft