LibreCAD/LibreCAD #1657: Circle and Ellipse has issues with different line types

app4soft, 2023-07-14:
### Expected behavior

All entities with the same line type should be drown as expected. 

### Observed behavior

Actually there are issue with line types applied to Circle and Ellipse:

1. Circle has much smaller dashes/dots/distance for all line types;
2. Ellipse has a bit broken look with `Dot` and other small dash/dot line types.

[image]
[image]
[image]
[image]
[image]

### Steps to reproduce or sample file

1. Create drawing with different entities;
2. Apply same style on all entities;
3. Repeat for all line types.

### Operating System and LibreCAD version info

```
Version: 2.2.1_alpha-231-g3974b220
Compiler: GNU GCC 11.3.0
Compiled on: Jul 14 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Debian GNU/Linux 12 (bookworm)
```

### Related issues

- https://github.com/LibreCAD/LibreCAD/issues/1476

dxli, 2023-07-14:
Considered the same issue as #1607 

Need to review all draw methods of basic geometries (line, arc/circle, ellipses, splines(quadratic and cubic)).

Line patterns should be defined in paper space. Current Qt painter supports lines/arcs/circles/complete ellipses, but not elliptic arcs or splines(Confirmation needed).

Another thought: port painter from LibreCAD-v3?

Help needed here.

app4soft, 2023-07-14:
> Considered the same issue as`#1607`

Not directly related to any of this:

1. Circles has different dash/dot/space sizes (smaller) then all other entities;
2. Ellipses has a bit broken look with all line types;
3. Splines with "No Pen" shown as "Continuous";
4. "Dash Dot" and "Dash Dot (small)" has a bit off-centered middle dots (an all objects, except Circle).

dxli, 2023-07-16:
Duplicate of #778

app4soft, 2023-07-22:
Update after 871768d

1. ~~Circles has different dash/dot/space sizes (smaller) then all other entities;~~ Fixed.
2. ~~Ellipses has a bit broken look with all line types;~~ Fixed.
3. Splines with "No Pen" shown as "Continuous" — LibreCAD crashes now, see 1662
4. ~~"Dash Dot" and "Dash Dot (small)" has a bit off-centered middle dots (an all objects, except Circle).~~ Fixed
5. For lines with non-zero line width Line patterns should be defined in paper space.
6. For lines with "`0.00mm`" width Line pattern should be defined in screen space (for printing it should be recalculated to paper space).
7. In "Draft mode" all lines Line patterns should be defined in screen space.

dxli, 2023-07-23:
paperscale support is still broken.

app4soft, 2023-08-01:
One more Line type issue - now on ***"Spline"*** entities: it has problems with any other then `Continuous` Line types (including with selection `Dash (tiny)` Line type).

Also, *"Spline"* looks like there are Line types applied two times same time on top of each other, but with opposite directions.

- DXF file: [seletcion-dash-tiny.zip](https://github.com/LibreCAD/LibreCAD/files/12234602/seletcion-dash-tiny.zip)

[image]
[image]
[image]

app4soft, 2023-08-03:
Another issue discovered: Line type's patterns mapped on `Polyline` and `Freehand` per segment, instead of continuous all along from first point to last point.

In same time, it mapped correctly on `Spline`, `Line` and `Spline through points` (i.e. continuously from first point to last point).

[image]

dxli, 2023-09-21:
drawing a reversed arc is now slower because of commit a6ab92b : the reason: QPainter or QPainterPath doesn't support reversed arcs well, so reversed arcs are now simulated by line segments.

closed 2024-05-16 by dxli