LibreCAD/LibreCAD #1658: Poorly anti-aliased Ellipses

app4soft, 2023-07-14:
### Expected behavior

Anti-aliasing on Ellipses should be the same quality as on Circles, Splines, PolyLines (Arcs & Lines).

### Observed behavior

Actually only on Ellipses anti-aliasing looks poorly.

### Steps to reproduce or sample file

1. Enable anti-aliasing;
2. Draw Circle, Spline and Ellipse;
3. Apply line width to wider (for better visibility of issue).

### Operating System and LibreCAD version info

```
Version: 2.2.1_alpha-231-g3974b220
Compiler: GNU GCC 11.3.0
Compiled on: Jul 14 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Debian GNU/Linux 12 (bookworm)
```

### Related issues

- https://github.com/LibreCAD/LibreCAD/issues/778

dxli, 2023-07-16:
Duplicate of #778

app4soft, 2023-07-22:
Fixed with 871768d

[image]

closed 2023-07-22 by app4soft