LibreCAD/LibreCAD #1670: Zoom is slower while a hatch is highlighted by the cursor

gerardbm, 2023-07-28:
### Expected behavior

Zoom in or zoom out with the mouse becomes slower than normally usage when the mouse is highlighting a hatch.

### Observed behavior

The zoom should be consistent and stable, independently of the viewer or the highlighted or selected objects.

### Steps to reproduce or sample file

Create a new file, create a polygon, add a hatch into it with a lot of lines. For example this one:

[image]

### Operating System and LibreCAD version info

Version: 2.2.1_alpha-255-gf4985ac4
Compiler: GNU GCC 11.3.0
Compiled on: Jul 28 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Debian GNU/Linux 9 (stretch)

dxli, 2023-07-31:
I am wondering whether we can move hatching to the display layer. Conceptually, nothing relies on the details of hatching lines or how they are trimmed exactly.

dxli, 2023-08-01:
Need to disable mouse hovering effects on hatch. Instead show borders

closed 2023-09-05 by dxli