LibreCAD/LibreCAD #1671: Selected object line type has low visibility

app4soft, 2023-07-28:
### Expected behavior

Selected objects line type should visible or there should be an option to customize "Selected Line type" in application preferences.

### Observed behavior

Actually selected objects almost invisible due to hard-coded "Dot" style.

[image]

[image]

### Steps to reproduce or sample file

1. Create some objects;
2. Select some object.

### Operating System and LibreCAD version info

```
Version: 2.2.1_alpha-255-gf4985ac4
Compiler: GNU GCC 11.3.0
Compiled on: Jul 28 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Debian GNU/Linux 12 (bookworm)
```

CTuBu, 2023-07-29:
Same on Win10

dxli, 2023-07-29:
Looks like the pen style of selected was lost to me

app4soft, 2023-07-29:
For me it looks like now exact `Dot` line type (which is in between `Dot small` and `Dot (large)`) is used as a line type for all selected objects. Previously selected objects has line type similar to `Dash (tiny)`.

Here is comparison (grid is off): on the left — selected spline with `Continious` line type, on the right — non-selected spline with `Dot` line type.

[image]

closed 2023-08-01 by app4soft