LibreCAD/LibreCAD #1683: Commands not working

grandixximo, 2023-08-15:
### Expected behavior

command is typed in command bar,
the tool is run as enter is pressed

### Observed behavior

Unknown command: xxxx 

is displayed 
where xxx is
fillet
offset
o
bevel

### Operating System and LibreCAD version info
Windows 11
2.2.1_alpha-276

abubkurian, 2023-08-15:
offset command is actually parallel, try pa; fillet is trim2 I think.

dxli, 2023-08-15:
We may printout a list of available commands (with short commands) when the user types "/?", "help", etc.

GSStnb, 2023-08-15:
A list of the commands can be shown in the 'Command line' dock by pressing [Tab].

grandixximo, 2023-08-16:
https://librecad.readthedocs.io/en/latest/ref/tools.html#tools
then need update the documentation, because those seems lik valid command according to the docs.
o it's a much quicker command tha pa or parallel, also they don't seem to call the exact same function, offset in the toolbar behaves differently from pa command.

kromanenko, 2024-02-03:
The aforementioned commands are invoked as expected in 
Version: 2.2.0.2
Compiler: Clang 13.0.0
Compiled on: Jul 29 2023
Qt Version: 5.15.10
Boost Version: 1.82.0
System: macOS 12.7

PeterHouseJr, 2025-01-22:
I just installed v2.2.1 on my win11 desktop (LibreCAD-v2.2.1-win64.exe).  Typing "line" and selecting two points draws a line as expected.  Then typing OFFSET results in "Unknown Command: offset" message.  Same for Fillet, Trim and probably some others. Using "lineoff" works as expected.  Clicking on the "Fillet" icon also works.  Selecting the "Offset" icon then prompts me to "Select to create offset" which allows me to select a line snf requires me to select one or more lines and press Enter or space to continue - when using lineoff I only have to hover over a line and hover on the side of the desired offset and click to complete (Very Nice BTW).

I have used AutoCad since version 1 (!) and am very command line oriented.  I have a previous version of LibreCad, 2.2.0.2 on my laptop which works as expected.

Please let me know if you need more information . . .

dxli, 2025-01-22:
@PeterHouseJr 

for command definitions, the latest 2.2.1 version is taking from the file:

https://github.com/LibreCAD/LibreCAD/blob/2.2.1/librecad/src/cmd/rs_commands.cpp

You can also define your command aliases, in the alias file also mentioned in the source code file:

 * In OS_WIN32 "c:\documents&settings\<user>\local configuration\application data\LibreCAD\librecad.alias"
 * In OS_MAC "/Users/<user>/Library/Application Support/LibreCAD/librecad.alias"
 * In OS_LINUX "/home/<user>/.local/share/data/LibreCAD/librecad.alias"

The alias file format is like:

> l line2p

A line like "l line2p" in the alias file will define an alias "l" to mean the "line2p" command.

PeterHouseJr, 2025-01-29:
Q - Is there a way to read in the alias file without restarting LibreCAD?

I looked over the RS_commands.cpp files for both v2.2.1 and my previous version of v2.2.0.1.  The differences are definitely coded into LibreCAD and it looks to me like AutoCad compatibility is no longer a requirement for LibreCAD and the original AutoCAD command aliases are no longer in the source code.  

This is understandable since there are probably fewer and fewer people who use the AutoCAD original command on the command line and they are using the two letter, newer, shortcuts.  

I have adapted personally by changing the alias file and thus not adapting!  This was easy since I only use the command line for about 25 commands and use the GUI for the rest which are used much less frequently.  The alias file was a very good idea whenever it was created.  

Q - I am curious why LibreCAD does not use the alias file for all command line conversions to drawing primitive commands?

For me, the file compatibility is the most important thing and that gets less important every year.

Thank you for your quick response.

Zopiac, 2025-04-19:
I've been struggling with this for a while now, myself. The loss of offset/o is my primary gripe, moreover when I select it from the Modify toolbar I get this message in the cmd box:

`Offset : offset, o`

implying that I can type either `offset` or `o` to attain the same effect. However, I get the `Unknown command` error on attempting either.

Furthermore, I tried to add an alias to `~/.local/share/data/LibreCAD/librecad.alias` as mentioned but upon relaunching LibreCAD none of the defined aliases seem to have worked (i.e. `o parallel` or even `il line2p`, just as a play on the builtin shortcut). I also found a precreated file in `~/.local/share/LibreCAD/LibreCAD/librecad.alias` instead and did the same there without success.

What am I doing wrong (other than needlessly relearning functions I've used for years)? At the very least, aliasing should work and/or the text echoed to the cmd window should properly reflect what the very same program expects you to use. After all, the menu shows Offset, and clicking it seems to inform me that either `offset` or `o` are valid verbs, without any reference to the term `parallel` or shorthand `pa`.

Bonus points for having tooltips enumerate the shorthand for selected command for better exposure... but that's extending past the scope of this issue.

dxli, 2025-04-19:
@Zopiac 
Try the latest build(2.2.2_aloha), and start librecad from a console. Find the alias file in the console output.

I think you can also hit tab in the librecad command input to get all commands and the alias file.

After closing any running librecad, you may backup and delete the alias file, and a new one will be generated at the next librecad startup.

To define a new alias, the alias is only required to be not an existing full command.

Zopiac, 2025-04-20:
> [@Zopiac](https://github.com/Zopiac) Try the latest build(2.2.2_aloha), and start librecad from a console. Find the alias file in the console output.
>
> I think you can also hit tab in the librecad [...]

Thank you, this did help the alias issue and helps restore prior functionality... although now it requires qt6ct which doesn't see the theme I have had selected in qt5ct, so it's kind of visually broken... well, again that's a different issue I suppose.

dxli, 2025-04-20:
@Zopiac 

After you fixed the alias in 2.2.2, you can switch back to 2.2.1.

The alias location is the same, but you probably need to delete the alias file and start from a fresh one again in 2.2.1.

Zopiac, 2025-04-20:
> [@Zopiac](https://github.com/Zopiac)
> 
> After you fixed the alias in 2.2.2, you can switch back to 2.2.1.
> 
> The alias location is the same, but you probably need to delete the alias file and start from a fresh one again in 2.2.1.

No dice. I'm still getting `Unknown command: o` on 2.2.1 after this. with `o parallel` defined in librecad.alias

guille2301, 2025-04-25:
Hello, I´m new user, for a few days, and I have had the same problems, after working with AutoCAD since version 11.
in my point of view I think that many commands have been renamed, for example 'offset' command, now is modoffset and isn't in shotcut list, I have added the sentence 'o	modoffset' in librecad.alias and now works fine, for me.

In line command, press [TAB] and you will have a complete list of commands, then tou can reconfigure your librecad.alias.

Sorry for my english.
Best regards.

pgmorgan, 2026-03-27:
I have submitted a PR #2513 with hopes that future releases can benefit from reverting to the historical CLI compatibility with AutoCAD that @grandixximo , @Zopiac , @PeterHouseJr , @guille2301 , and I chimed in about.

Note that as a CMD (CLI) user principally, I have not reviewed how changes in the CLI map to the GUI buttons.

If others can review my PR and/or test this it would be appreciated.

pgmorgan, 2026-05-01:
I am an educator and an advocate for `librecad`.  This is a critical issue for me, my current and future students, and others that have chimed in.  I just put [a thought-out comment](https://github.com/LibreCAD/LibreCAD/pull/2513#issuecomment-4358035560) on PR #2513 as I hope to advance this resolution for the community at large.

It is copied here for community members that may find or follow the issue rather than a specific PR to address it.

-------------------------------

@sand1024 
@dxli 

Thanks for your comments.  Whether with @sand1024  suggestion or not, I still think the **default CLI** commands should **revert** to at least include the cross platform compatible CLI commands.  

The change away from platform agnostic CLI compatibility is very recent.  It happened between 2.2.0 and 2.2.1.

It is important to look at 2D CAD drafting as a skill, and not only a platform specific tool.  AutoCAD has 43 years of decent CLI stability.  LibreCAD has benefitted from sharing that stable CLI for 15 years.  This has allowed people to learn and apply this skills across tools and computers with little friction.  The LibreCAD community of users can benefit from returning to that common CLI as it facilitates access and application of the skill in a broader set of computational and institutional requirements at home, at school, at work, and across computers with greater or lesser compute resources.

As a critical case in the Linux community (which has a wide diversity of types of users), it is very common to access software in the package manager repositories and there are benefits to this approach even if it delays access to releases.  All current LTS Ubuntu versions and all current LTS Debian versions, including variant distros based on these, are still on `librecad` 2.2.0 or earlier in the `apt` repositories.  When such users who access `librecad` through their package managers arrive to version 2.2.1 or greater in the future, the `librecad` community would have long bifurcated along this critical and unnecessary interface difference unless we revert this change.

As an advocate for both `librecad` and FOSS I think this should not be overlooked, and should instead (with charity and without blaming) be treated as a _regression_.  I don't think it should be left to each user to manually re-align this common compatibility individually, especially given the reasonable delay in the user base reaching current releases and the benefits of a cross tool CLI.

Here are the changes I have identified to revert this:

```
incompatible command --> compatible command

lineoff    --> offset
modrotate  --> rotate
modscale   --> scale
modmirror  --> mirror
modtrim    --> trim
modoffset  --> offset
modbevel   --> bevel
modexplode --> explode
```
Other critical commands maintain good compatibility like `l`, `pl`, `c`, `move`, `mtext`, `dimlinear`, and many other common commands.  But the above list is the scope of the regression I have identified.  I hope that we can clean this up ourselves rather than leave it to each future user to handle or not.

Thank you kindly

dxli, 2026-08-25:
PR #2513 has been reworked around a simple rule: the most reasonable/cross-platform name becomes the main command for its action, and whatever it displaces becomes a short command on the same action - so nothing that works today stops working, and the AutoCAD-compatible names (`rotate`, `scale`, `mirror`, `trim`, `explode`, `fillet`, `bevel`, `offset`, `parallel`, plus `chamfer`) come back as main commands.

One exception worth flagging here since this issue is where the original renames were discussed: `offset` stays bound to the actual Modify > Offset tool rather than the parallel-line tool it pointed to before 2.2.1. That's deliberate - see #1833 for why - and it's the one place "restore the old behavior" and "make the CLI work correctly" pointed in different directions.

Leaving this open until #2513 actually merges.

dxli, 2026-08-25:
Fixed by PR #2513 (merge a05b4261f), which restores rotate, scale, mirror, trim, explode, fillet, bevel, offset and chamfer as main commands while keeping the mod*-prefixed 2.2.1 names working as short commands.

closed 2026-08-25 by dxli