LibreCAD/LibreCAD #1684: Wrong position of text

bansan85, 2023-08-16:
Texts are wrongly placed in master.

[plans_2023_07_21.zip](https://github.com/LibreCAD/LibreCAD/files/12360509/plans_2023_07_21.zip)

### Expected behavior

Open enclosed file with

Version: 2.2.0.2
Compiler: GNU GCC 13.1.0
Compiled on: Jul 29 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

[image]

### Observed behavior

Open enclosed file with

Version: 2.2.1_alpha-278-g5064b9ca
Compiler: GNU GCC 13.2.0
Compiled on: Aug 16 2023
Qt Version: 5.15.2
Boost Version: 1.75.0
System: Windows 10 Version 2009

[image]

PS : My windows system is Windows 11 version 22H2 not "Windows 10 Version 2009" (which is probably the one from Github CI).

dxli, 2023-08-16:
It's most likely a regression due to right-to-left text support: PR #1674 

Probably a duplicate of Issue #1675

dxli, 2023-10-20:
Thanks for reporting. The regression was indeed a tricky one passed my code review.

closed 2023-10-20 by dxli

bansan85, 2023-10-20:
Thanks, I just tested and every thing looks fine. Silly virtual function ;-)