LibreCAD/LibreCAD #1687: greek fonts problems

jimishol, 2023-08-28:
### Expected behavior
Greek characters should be used out of the box.

### Observed behavior
I had problems writing greek by MText. Some fonts displayed diamonds instead and a font was partially succeeded.
Probably there is a problem with the keyboard layout greek people use.

### Operating System and LibreCAD version info
Opensuse tubleweed and librecad 2.2.0-1.4

Anyway, as i had mentioned in https://forum.librecad.org/greek-fonts-large-file-from-dwg-to-librecad-td5714230.html and in https://forum.librecad.org/Insert-text-in-greek-language-td5717563.html, i had solved the problem by editing the greeks.lff and unicode.lff files. I checked today and these files works fine, as long as i define their paths in preferences. Feel free to use them as you wish. I attach these files I edited. I didn't touched first lines, so Hershey Rallaz is mentioned as author of these as he should be.
[greeks-mine.lff.txt](https://github.com/LibreCAD/LibreCAD/files/12455550/greeks-mine.lff.txt)
[unicode-mine.lff.txt](https://github.com/LibreCAD/LibreCAD/files/12455552/unicode-mine.lff.txt)

(Obviously, remove .txt file extension)

dongxuli-concord-design, 2023-08-28:
Can you submit a minimum dxf to show this issue?

jimishol, 2023-08-28:
[mine_fonts.dxf.txt](https://github.com/LibreCAD/LibreCAD/files/12456341/mine_fonts.dxf.txt)
[not_mine_fonts.dxf.txt](https://github.com/LibreCAD/LibreCAD/files/12456342/not_mine_fonts.dxf.txt)
In these dxf files appears all greek letters that my keyboard produce naturally.
not_mine_fonts is created my librecad without setting special font path. On my laptop neither unicode.lff fonts can make them appear (at least in preview).
mine_fonts is the same dxf after configuration of font path to use unicode-mine.lff.  These time all possible greek letters appear properly in preview at my laptop.

lordofbikes, 2023-08-28:
Basically we have an issue with UTF/Unicode and ASCII codepage mapping.
The `greek*.lff`files contain only 8-bit ASCII code characters. They may work on non-UTF/Unicode systems, where the keyboard creates 8-bit codes for a defined system codepage.

Your `unicode-mine.lff` adds the missing greek UTF-8 characters, `U0386` to `U03d4`, that's why this font works.

Sadly the original `unicode.lff` has got some improvement, so we can't simply pull yours into the repo. But your additions are clear and I can merge them manually into the latest `unicode.lff`.

Concerning the other `greek*.lff` fonts, we can probably apply the same changes from `unicode.lff` to make them work properly.
I just can't understand your changes in `greeks-mine.lff` yet, it needs a deeper look if there were some improvements in the deployed `greeks.lff` font too.

jimishol, 2023-08-28:
> I just can't understand your changes in greeks-mine.lff yet,...

Today, i cannot remember how I managed to edit lff files. What i remember is that I firstly edited the greeks.lff file. greek fonts were there but corresponded to wrong keys for my keyboard layout. So, probably i used xev program and just rearranged fonts to my keyboard layout. The only true design from scratch was letter "ς" that corresponds to "w". After, that i merged the already existed greek letters to unicode.lff accordingly. 
Unfortunately, that is all I remember.