LibreCAD/LibreCAD #1689: Open Spline Requires 4 Points for 3-point Spline

flywire, 2023-08-31:
### Expected behavior

An open spline should close on three points.

### Observed behavior

An end point must also be entered as a control point to close a 3-point spline.

### Steps to reproduce or sample file

```
spl
Command: spl (spline)
30,0
30,40
0,40
0,40
close
Unknown command: close
(right click in gui to close)
```
### Operating System and LibreCAD version info

Version: 2.2.0.1
Compiler: GNU GCC 13.1.0
Compiled on: Jul  8 2023 (interim build)
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

dxli, 2023-09-06:
For open (non-closed) cubic splines, 4 control points are still needed

flywire, 2023-09-06:
> For open (non-closed) cubic splines, 4 control points are still needed

Can you explain? The example provided has the last end point of the open spline on the previous control point so it seems it should work with two end points and one control point.

dxli, 2023-09-07:
RS_Spline is based on the standard version of B-spline, and the basis function order determines minimum size of the knot vector.

The basis function is linear is the knot vector is 2, and each additional knot value increases the basis function order by one.

The code is in rs_spline.cpp ( function: rbasis() ).

The math is described in
https://en.wikipedia.org/wiki/B-spline

flywire, 2023-09-07:
Thanks, that's great.

I'm not up to the C++ code but I've hacked [this demo](https://jsfiddle.net/flywire/ph4cfavd/):

[image]

```js
var brd = JXG.JSXGraph.initBoard('jxgbox',{boundingbox:[-10,50,40,-10],keepaspectratio:true,axis:true});

var p = [], col = 'red';
p.push(brd.create('point',[30,0],{strokeColor:col,fillColor:col}));
p.push(brd.create('point',[30,40],{strokeColor:col,fillColor:col}));
p.push(brd.create('point',[0,40],{strokeColor:col,fillColor:col}));

var c = brd.create('curve', JXG.Math.Numerics.bspline(p,4),
               {strokecolor:'blue', strokeOpacity:0.6, strokeWidth:5});
```

flywire, 2023-09-07:
Hmm, the demo also calculates a line for 2 points so clearly they are using whatever they can to interpolate between points. It seems reasonable to extend spline to use quadratic for 3 points.

| Points | Interpolation |
| ------ | ----------- |
| 0 | None |
| 1 | Point |
| 2 | Line |
| 3 | Quadratic |
| 4 | Cubic |

dongxuli-concord-design, 2023-09-07:
Can you confirm it is not the same as quadratic spline with 3 points?

Our engine is using uniform cubic b-spline, so we cannot use degenerate cubic.

SplinePoints is exactly quadratic, can be used for comparison

flywire, 2023-09-08:
I don't understand the concepts well enough to understand your communication so I can't confirm anything. 

https://jsxgraph.org/docs/symbols/JXG.Math.Numerics.html#.bspline

> Computes the B-spline curve of order k (order = degree+1) through a given set of points.

This is an unreasonable claim if a B-spline curve requires a minimum of four points and the class calculates for two or three points.

I've also tested the class visually by repeating the first or third point for a total of four points and it gives slightly different curves, both of which are very different from the curve generated for three points. While repeating a point was adequate for my use-case it seems it is only a kludge.

flywire, 2023-09-11:
[ Received by email ]

> Could you guide us with the 3-point open B-spline implementation in https://librecad.org/ ? LibreCad currently requires a minimum of 4 points but https://jsxgraph.org/docs/symbols/JXG.Math.Numerics.html#.bspline works with 3 (or even 2).

As far as I remember, we reduce the degree of the spline in case the number of points is not sufficient. See https://github.com/jsxgraph/jsxgraph/blob/main/src/math/numerics.js#L2806-L2808

Best wishes,
Alfred Wassermann

dxli, 2023-09-11:
Our spline code v2 is old and messy, so it's risky to add new features there. It might be easier to refactor and reduce technical debts first to add features (like, degenerate to lower orders, or non-uniform or degenerate knot vectors).

For b-splines, we will probably use quadratic for 3 points, then, it's the same as our SplinePoints. Also, conceptually, our SplinePoints can be tweaked to support all conic sections from 3 points. This direction would gives us the long time requested features: hyperbola and parabola.

Another direction could be adding GUI support for knot vector editing.

For the time being, if you want non-closed spline by 3 points, SplinePoints is the way to go, also support for SplinePoints is rather complete, compared with spline.