LibreCAD/LibreCAD #1691: Ortho lines get slow if extremely zoomed in

porentief49, 2023-09-18:
Orthogonal lines somehow get very slow when zoomed into a drawing. I'm using drawings that have a complete house, but will on the same page draw out some details in milimeters. When I do this, orthogonal lines can take several seconds to show the preview on any snap point. I see the same regardless of what snap is active.

### Expected behavior

Performance should not depend on the zoom factor. It's great under normal conditions, and I think it should be the same when zoomed in.

### Observed behavior

Depending on the zoom factor, ortho line preview gets slower and slower. That may be several seconds or more. At first I thought the app has crashed, but it only freezes and then comes back.

### Steps to reproduce or sample file

Open a new drawing and draw any line. Then pick the ortho line tool, select that line and try to place an orthogonal line. Now zoom in to crazy levels. Observe how it gets super-laggy.

### Operating System and LibreCAD version info

***Dell XPS 15 7590***
Device name	peng
Processor	Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz   2.40 GHz
Installed RAM	32.0 GB (31.7 GB usable)
Device ID	F375FA37-9A42-4716-A1C4-22DCA0996719
Product ID	00330-50000-00000-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Pen and touch support with 10 touch points
Screen Resolution 4k (3840 x 2160)

***Windows 11 Pro***
Version	22H2
Installed on	‎2022-‎11-‎09
OS build	22621.2283
Experience	Windows Feature Experience Pack 1000.22662.1000.0

***LibreCAD***
Version: 2.2.0
Compiler: GNU GCC 12.2.0
Compiled on: Dec 17 2022
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

dxli, 2023-09-20:
I cannot reproduce this issue.

Could you upload a sample dxf file and detailed steps to reproduce this issue?

lordofbikes, 2023-09-20:
This issue is about bad performance during orthogonal line creation, so a DXF does not help.
This made it reproducible for me on Linux:
- start a new drawing
- make a random diagonal 2-point line
- use Orthogonal line tool
- select the created line
- now, with the floating orthogonal line zoom in and see the performance loss

To be honest, you have to zoom in really deep to see the difference.
When the grid is 10 units on line creation, I had to zoom in until grid is in the hundredth fraction to see the effect. Means a zoom factor of 1000 or even 10000.

We know about other issues, e.g. in surveying, where these different levels of details create issues with accuracy and math, but this one is new to me.

dxli, 2023-09-21:
No, I cannot reproduce this one.

I tested zooming all the way to the hard limit: factor.x=1.93734e+07.

I did see some drawing artifacts in the process, but there was no performance issue.

porentief49, 2023-09-21:
Interesting, as it reproduces for me with both the original drawing I was working on and with a new one as @lordofbikes described. I'm attaching my plan for reference

[Plan.zip](https://github.com/LibreCAD/LibreCAD/files/12680836/Plan.zip)
hing my drawing for reference.

You'll see my complete neighborhood scaled to meters as I received it from the land registry office. In there, I'm drawing details of things that may be as small as single digit millimeters. So yes, a zoom of ~10k can happen here:

[image]

lordofbikes, 2023-09-21:
This issue exists in release versions 2.2.0 and latest 2.2.0.2, but not in latest master.
So this was solved by accident in the `master` branch.
I consider to bisect to find the fixing commit and apply it for a possible future bug fix release.

lordofbikes, 2023-09-21:
Or maybe it's related with different Qt versions? Then it may not be backported.

porentief49, 2023-09-22:
cool. Let me know when you have a release I can test this out to confirm. Thanks guys for the work you're putting in here. I would consider helping, but I'm mostly a .NET guy and a little scared when it comes to pointers ...

dxli, 2023-09-22:
@lordofbikes The problem was due line pattern rendering. When rendering a line with line style/pattern, the pattern is created in  in RS_Line::draw() in v2.2.0.2. The patterns a in the view/paper space, so when zoomed in, more pattern lines will be drawn. When thousands or more line segments are drawn, it becomes a performance issue.

This is a regression caused by commit 0e167183

The line patterns are only drawn within the viewport rectangle, so line segments are not generated outside of the view port. When zoomed in, only a tiny portion of the line is within the viewport. Commit 0e167183 commented the code in RS_Line::draw() doing intersection check with the viewport.

In the latest master, we stopped generating line patterns from our side, instead the line patterns are all handled by Qt (QPen), and Qt has its own viewport check to avoid generating huge number of invisible line segments outside of the viewport. Therefore, the issue doesn't exist in the master HEAD.

Backporting the whole rendering work to 2.2.0.x is not the best choice though. Reverting part of  0e167183 on the viewport bounding box check in RS_Line::draw() seems to be a safer way to go.

closed 2023-10-06 by dxli