LibreCAD/LibreCAD #1692: Mac Release doesn't launch on Mac OS 10.13

elosha, 2023-09-23:
I'm trying to launch the recent build of 2.2.0.2 for Mac OS, but the program crashes on launch. Guess this is because I'm on Mac OS 10.13 on my old MacPro5,1. Yet it would be cool to either a compatible version (optional) and a defined minimum OS requirement (both in README.md and in info.plist) 😊

### Expected behavior
Program launches.

### Observed behavior
Program crashes on launch due to incompatible library version, a symbol is missing.
```
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x4] Symbol missing

Dyld Error Message:
  Symbol not found: ____chkstk_darwin
  Referenced from: /Applications/LibreCAD.app/Contents/MacOS/../Frameworks/libpcre2-16.0.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib
```

### Steps to reproduce or sample file
Try to run pre-built LibreCAD release on Mac OS 10.13.x

### Operating System and LibreCAD version info
Mac OS 10.13.6 "High Sierra" for Intel
LibreCAD pre-built Mac release 2.2.0.2
Also tested 2.1.x

dxli, 2023-09-27:
I suppose you can still build from source, following the wiki:

[Build on OSX](https://librecad.readthedocs.io/en/latest/appx/build.html#building-on-macos)

If you run into dependency issue (Qt version), try to build from the latest git, instead of the 2.2 branch.

elosha, 2023-09-27:
@dxli Thanks. I installed it using MacPorts, however the binary releases here need adjustments ;)

dxli, 2024-05-18:
With 2.2.1 release candidates available, closing this one

closed 2024-05-18 by dxli