LibreCAD/LibreCAD #1693: Nonsensical Scroll Limits??

nanashi-1, 2023-09-27:
When I started this drawing scrolling worked fine. Closing the program and opening the file again does this weird scroll behavior. When I scroll to the right it stops me even when I haven't reached the end of the drawing.

**Left Most Horizontal Scroll**
[image]

**Right Most Horizontal Scroll**
[image]

### Sample file

https://mega.nz/file/mUQ2UaJK#eQPSr2oWFFGZhhvNZUpXA95kd_WZHC1AB76SwmR_xMI
*Note: I cannot upload this file type here in GitHub*

### Operating System and LibreCAD version info

#### System Details Report
---

##### Report details
- **Date generated:**                              2023-09-27 17:58:18

##### Hardware Information:
- **Hardware Model:**                              ASUSTeK COMPUTER INC. VivoBook_ASUSLaptop X515JAB_X515JA
- **Memory:**                                      8.0 GiB
- **Processor:**                                   Intel® Core™ i7-1065G7 × 8
- **Graphics:**                                    Intel® Iris® Plus Graphics (ICL GT2)
- **Disk Capacity:**                               (null)

##### Software Information:
- **Firmware Version:**                            X515JAB.308
- **OS Name:**                                     Arch Linux
- **OS Build:**                                    rolling
- **OS Type:**                                     64-bit
- **GNOME Version:**                               45.0
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.5.3-arch1-1

##### LibreCAD Information:
-**Version:**             2.2.0
-**Branch:**             stable
-**Installation:**     system
-**Provider:**          flathub

dxli, 2023-10-04:
I feel this is by design:

the scroll bars only cover about 3 times of the current viewport, to avoid super fast scrolling, if zoomed in factor is huge.

lordofbikes, 2023-10-04:
The provided file contains some weird extension limits `$EXTMIN`/`$EXTMAX` of +/- 10000000000.

First, this only happens after opening the provided file, using any kind of Zoom (tool buttons or mouse wheel) fixed the scroll bars for me.
Using Auto Zoom and save the file again fixed the extension limits.

So the question here is, where these `$EXTMIN`/`$EXTMAX` values come from, and why they irritate the scroll bars on initial load.

nanashi-1, 2023-10-05:
> First, this only happens after opening the provided file, using any kind of Zoom (tool buttons or mouse wheel) fixed the scroll bars for me.

Weirdly, using any kind of Zoom doesn't fix my issue.

> Using Auto Zoom and save the file again fixed the extension limits.

This fixes it.

lordofbikes, 2023-10-05:
No matter how these `$EXTMIN`/`$EXTMAX` values were saved, they should not make portions of the drawing unreachable on opening.
Auto Zoom is a work around, but not the solution for this issue.