LibreCAD/LibreCAD #1694: Revert Russian translations from gfghjjk

dsa-t, 2023-09-30:
This user is adding terrible machine-generated translations to many open-source projects via Weblate: https://hosted.weblate.org/user/xXx/#changes or @gfghjjk

Most importantly, they're replacing correct translations without understanding the terminology or context.

These changes should be reverted.

dongxuli-concord-design, 2023-09-30:
Wondering what's the motivation for this type of vandalism.

dsa-t, 2023-09-30:
I wouldn't say it was with malicious intent, but:

Looking at their comments, they don't understand basic rules of the Russian language:
https://hosted.weblate.org/changes/?user=xXx&action=3

They don't understand that markdown links shouldn't be translated:
https://hosted.weblate.org/translate/klipper/rotation_distance/ru/?checksum=7681c4c3cba45133

They don't understand that config file examples shouldn't be translated:
https://hosted.weblate.org/translate/klipper/rpi_microcontroller/ru/?checksum=1d51ac69e8f8e03b

Or command line output example:
https://hosted.weblate.org/translate/klipper/rpi_microcontroller/ru/?checksum=439eae00602202ee

Or command line flags `./scripts/flash-sdcard.sh -h` -> `./scripts/flash-sdcard.sh -ч`
https://hosted.weblate.org/translate/klipper/sdcard_updates/ru/?checksum=1e09cfa71a824563
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=0a14d2c600dfc83e

Similar with Lektor attributes:
https://hosted.weblate.org/translate/tor/tpo-web/ru/?checksum=c61a47021b890db7

Fluent attributes:
https://hosted.weblate.org/translate/tor/tor-browser/tb-tor-browser-ftl/ru/?checksum=4decf2af9773cfd6
https://hosted.weblate.org/translate/tor/tor-browser/tb-tor-browser-ftl/ru/?checksum=67cfd749485e9bb9

Messing up LinuxCNC terminology, destroying correct translations:
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=c8827bd1480a80dc
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=95ffef34a5c0f75e
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=8f92d6ed06943e7d

Not sure why it is translated, but it's a completely robotic translation:
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=def0dab2e88826da

Not even they care slightly about any kind of quality:
https://hosted.weblate.org/translate/librecamera/librecamera/ru/?checksum=138ea8113804ec72
https://hosted.weblate.org/translate/librecamera/librecamera/ru/?checksum=c2791bec24f98ac2

Now, in LibreCAD:
**They literally translated "linetan2c" as "suntan line2c":**
https://hosted.weblate.org/translate/librecad/librecad/ru/?checksum=6186e9808a22d4b1
And that's just one example

This command "линия, разделяющая пополам" probably won't even work (I don't use LibreCAD myself):
https://hosted.weblate.org/translate/librecad/librecad/ru/?checksum=48dd49dd7c8aacd0

Broken C format:
https://hosted.weblate.org/translate/librecad/librecad/ru/?checksum=968d9a7c9adfe75c
https://hosted.weblate.org/translate/udplogger/udplogger/ru/?checksum=8a43581bc4912e1a
https://hosted.weblate.org/translate/circuitpython/main/ru/?checksum=e526aff56e7f356b

Doesn't understand what & means, thinks this is an error. For info: & specifies mnemonics/Access keys (Alt + <character>).
https://hosted.weblate.org/translate/librecad/librecad/ru/?checksum=564bd8778cabdbfa

"GroupCrate":
https://hosted.weblate.org/translate/librecad/librecad/ru/?checksum=61c99edc426f818e

Translating program names:
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=a2ad5d9314416fe7
https://hosted.weblate.org/translate/linuxcnc/linuxcnc/ru/?checksum=b188a4109d00aecb
https://hosted.weblate.org/translate/tor/community-portal/ru/?checksum=0c8272a220eca1c8

Broken rst markup in Weblate docs:
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=90acae8d5d084762
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=20f6e51d0008f7d8
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=2273a1b0f8179af2
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=2e8c2def84c741c6
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=880e862717e0dd13
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=1d79a93b4d17940f

Key strings in Weblate docs that shouldn't be translated:
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=28493981cb5b187d
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=9e22cbca05987424
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=305721a4b222b035
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=7d01a7636ac274c9
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=f0a14f1c2542349f

CircuitPython types that shouldn't be translated:
https://hosted.weblate.org/translate/weblate/documentation/ru/?checksum=88c816efd747505b

closed 2026-09-26 by sand1024