LibreCAD/LibreCAD #1695: Cannot select arrowhead of exploded dimension

carlcom, 2023-10-03:
Because of issue #1512 I explode a dimension, after positioning the text, to keep its position. 

The color gets changed when exploded, even though it is still the dimension layer. I modify the text and line color, but I cannot select (left mouse click) an arrowhead when "Click on entity to modify". 

### Expected behavior
 Clicking arrowhead brings up its properties to modify.

### Observed behavior
 Clicking arrowhead does nothing

### Steps to reproduce or sample file "We don't support that file type." dxf file. Explode a dimension, then try to change the arrowhead color.

### Operating System and LibreCAD version info
Windows 11 LibreCAD 2.2.0.2

dxli, 2023-10-04:
Can you area select the arrowhead?

I have no trouble on selection of exploded dimensions, but I'm running the latest master branch.

carlcom, 2023-10-04:
If I understand area select, I cannot use that to select with Modify Properties.

dellus, 2023-10-04:
Confirmed, the exploded arrowhead does not react to Properties.
As a workaround, simply use Attributes instead. This works, and you even can select multiple entities, convenient.

dxli, 2023-10-04:
I got it. It's about properties. In the master branch, double click should open the property dialog.

Yes, I can confirm the arrow head doesn't support this feature. I will try to understand why. Currently, the hatch property dialog only allows changing the attributes, so as @dellus suggested, try Attributes instead.