LibreCAD/LibreCAD #1703: PDF export defaults with margin

PeterHagen, 2023-10-17:
### Expected behavior

When exporting to a PDF, with margin's set to 0, the entire drawing should be visible on the exported page. This works as expected in 2.2.0.2.

### Observed behavior

With version Version: 2.2.1_alpha-301-gd3985b6c (and earlier, but lost track), a margin is always added to the exported PDF

A print preview:

[image]

The exported pdf:

[image]

### Steps to reproduce or sample file

1. open the dxf
2. check the print preview
3. export to pdf

[test.pdf](https://github.com/LibreCAD/LibreCAD/files/12941993/test.pdf)
[test.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/12942090/test.dxf.zip)

### Operating System and LibreCAD version info

MacOS 14.0 (23A344)
LibreCad 2.2.1_alpha-301-gd3985b6c (build from source for M1)

PeterHagen, 2023-12-17:
Does anyone have an idea how to solve this? This bug prevents me to use any new version of Librecad

dellus, 2024-01-15:
I have no knowledge about MacOS, but for me in Windows this happens as automatically the physical print margins of the default printer are applied. To circumvent this I have a pdf - printer (just software) set as default printer. If there is something like that for MacOS you could try that.

PeterHagen, 2024-01-15:
That could be a solution, but then the implementation of the export to pdf is not what I expect it to be. Export to PDF should not use settings of a printer, it should create a file that is independent of any printer. That is what PDF is for.

It does work in the older version 2.2.0.2, which is an Intel version. I would prefer to use the Arm version, but some bugs and specially this export issue restrain me of using them.

dxli, 2024-05-18:
Workaround available

closed 2024-05-18 by dxli