LibreCAD/LibreCAD #1706: Margin to select element to create a parallel line doesn't scale with the view

bufalo1973, 2023-11-01:
### Expected behavior
The margin to select an element to make a parallel line should be in screen pixels.

### Observed behavior
When you zoom out, the margin gets narrower and ends up being one pixel (if you're lucky)

### Steps to reproduce or sample file
Just try creating a parallel line when the view is above the thousand units wide.

### Operating System and LibreCAD version info
version 2.1.3 on KDE Neon (up to date).

lordofbikes, 2023-11-01:
This sounds like a duplicate of #1560.
It is still not reproducible for me.
If you can share a sample file, or more details, please post it in #1560 (rename or zip the DXF for GitHub).

Anyhow, we will not fix this for 2.1.3, our current release is 2.2.0.
You can try the latest release AppImage or the continuous build from here:
https://github.com/LibreCAD/LibreCAD/releases

closed 2023-11-01 by lordofbikes

bufalo1973, 2023-11-01:
I know 2.1.3 has ended support. I posted this in case it happens also in 2.2.

And the problem arises when you zoom out a lot (a square around 10000 mm) and try to make a parallel line.

If the problem is solved in 2.2, great.

lordofbikes, 2023-11-01:
#1560 is about 2.2.0-rc4, I think the last release candidate before the final release.
There were no commits addressing this issue, so it is probably still there in latest version.
But we got no sample file to see if it is reproducible and contained in the DXF file. I tried again, but I don't have this issue with my setup. So I have different settings or I'm missing something when I try to reproduce it.

bufalo1973, 2023-11-02:
It's strange. I've repeated the steps to reproduce it but now it works right.