LibreCAD/LibreCAD #1710: Share experience using ttf2lff to convert Traditional Chinese lff font

purplewish07, 2023-11-17:
I found that almost no one discussed the Traditional Chinese font issue.

So I share my experience.

I downloaded the Noto Sans Traditional Chinese Font from [Google Fonts](https://fonts.google.com/).

And use ttf2lff tool to convert NotoSansTC-Thin.ttf to notosantc-thin.lff
`ttf2lff <ttf file> <lff file>`

I found the llf file name must be lowercase otherwise it will display a diamond when you use unicode words.

Copy notosantc-thin.lff to C:\Program Files (x86)\LibreCAD\resources\fonts

Now I can use notosantc-thin.lff in LibreCAD.

By setting the Word Attributes Pen Width to 0.13mm, the text appears very similar to a single line font.
[image]

I'm very happy to have successfully resolved the issue with the Traditional Chinese lff font.

lordofbikes, 2023-11-17:
Many thanks for sharing 👍 
As you mentioned, this issue is solved, so I close it.
It is already shared in the forum, what is a better place for sharing experiences.
https://forum.librecad.org/How-to-use-Chinese-drawing-in-this-software-tp5723075p5723872.html

This can also be a good improvement for our fonts wiki article.
https://dokuwiki.librecad.org/doku.php/community:fonts
Feel free to register there and add your experience to the fonts article.

closed 2023-11-17 by lordofbikes

lordofbikes, 2023-11-17:
> wenquanyi fonts

Is available in the wiki https://dokuwiki.librecad.org/doku.php/community:fonts#additional
The intention is to add fonts there, when their licenses allow conversion and public sharing.