LibreCAD/LibreCAD #1717: Looking forward to Qt6 Compatability...

jiapei100, 2023-12-10:
Looking forward to Qt6 Compatability...

### Expected behavior

Compatible with Qt 6.6.1

### Observed behavior

Failed to build from source for now

### Steps to reproduce or sample file

Just build with Qt 6.6.1, for instance:

```console
lib/filters/rs_filterdxfrw.cpp:28:10: fatal error: 'QTextCodec' file not found
   28 | #include <QTextCodec>
      |          ^~~~~~~~~~~~
lib/filters/rs_filterjww.cpp:27:10: fatal error: 'QTextCodec' file not found
   27 | #include <QTextCodec>
```

### Operating System and LibreCAD version info

```console
➜  ~ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
```
with LibreCAD **2.2.0.2**...

Cheers

dongxuli-concord-design, 2023-12-10:
We have a qt6 branch at GitHub. It should build at least.

Please help improve qt6 support by using it.

jiapei100, 2023-12-10:
@dongxuli-concord-design 

Successfully built... Thank you ...

jiapei100, 2023-12-10:
But, where is the installation???  
After my installation, I only found:
```
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/librecad
/usr/share/doc/librecad/LICENSE
/usr/share/doc/librecad/README.md
``` 
are installed. Where are those **libraries** and **executables** that should also be installed????

dxli, 2024-03-30:
How did you install?

https://docs.librecad.org/en/latest/appx/build.html

For linux, do the following in the LibreCAD source code top folder.

```bash
qmake6 -r
make distclean
qmake6 -r
make -j8
unix/librecad
```

dxli, 2024-05-16:
We have continuous build on Qt6 for the master branch now.

Closing this as fixed

closed 2024-05-16 by dxli