LibreCAD/LibreCAD #1718: Arc Tangential command options problem

akcansoft, 2023-12-10:
When the values in the Arc Tangential command options are changed, they revert to the previous values.
It automatically switches to the Angle option when the Radius option is selected
or
it automatically switches to the Radius option when the Angle option is selected.
----
Version: 2.2.1_alpha-319-gf4720a0f
Compiler: GNU GCC 13.2.0
Compiled on: Dec  6 2023
Qt Version: 5.15.2
Boost Version: 1.75.0
System: Windows 11 x64

akcansoft, 2024-01-06:
The problem with this command has been solved but there are similar problems with other curve command options.
I publish Librecad tutorial videos on YouTube. I experienced such problems in my last video.
https://youtu.be/5RQ-qHbavIU

Moments in the video when the problem occurs:
https://youtu.be/5RQ-qHbavIU?t=127
https://youtu.be/5RQ-qHbavIU?t=339
https://youtu.be/5RQ-qHbavIU?t=382
https://youtu.be/5RQ-qHbavIU?t=440

System: Windows 11 TR
Librecad Version: 2.2.1_alpha-326-g5c8300ed
Compiler: GNU GCC 13.2.0
Compiled on: Jan  3 2024
Qt Version: 5.15.2
Boost Version: 1.75.0

dxli, 2024-02-29:
We should fix similar issues now and prepare for a new release.

dxli, 2024-04-02:
Fixed

closed 2024-04-02 by dxli