LibreCAD/LibreCAD #1720: Applying the color of the layer to the block does not work

kberkut, 2023-12-31:
### Expected behavior
Changing the color of the layer should result in a change in the color of the block lines

### Observed behavior
Changing the color of the layer does not affect the color change of the block lines

### Steps to play or select a file
Insert any shape from the block library.
Move the shape to another layer with a different layer color.

### Information about the operating system and version of LibreCAD
Version: 2.2.0-undef
Compiler: GNU GCC 13.2.1
Compiled on: Nov  7 2023
Qt Version: 5.15.11
Boost Version: 1.83.0
System: Arch Linux

OS: Arch Linux x86_64 
Kernel: 6.6.8-arch1-1 
DE: GNOME 45.2 
CPU: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (8) @ 2.100GHz 
GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series 
Memory: 4264MiB / 9884MiB

lordofbikes, 2023-12-31:
Works for me, e.g. with misc/tux block from deployed libraries.
Insert to layer 0 results in white lines, move block to layer 1 (green), results in a green tux.

You should check the attributes of the block entities in the block editor.
When the color is set fixed, it does not change to the layer attributes.
I'm not sure what `BY BLOCK` color attribute will give.
But for the expected color change the block entities color should be set to `BY LAYER`.

kberkut, 2023-12-31:
In the properties of the blocks, everything is set by layer. But the color is not applied.

dongxuli-concord-design, 2023-12-31:
Our block features need a code review anyway. I am too busy now to allocate some solid time here.

It's probably the last item for a new stable release. After that, we should switch to qt6.

kberkut, 2024-01-01:
I found out some special feature. Colors are not applied to embedded libraries installed with LibreCAD. But they are used for user-connected libraries that are specified in the settings.

rola22, 2024-01-12:
Hi.
I am currently on LibreCAD 2.2.0.2-5-g3aa2c41c and everything works like in commercial CAD software:
- if geometry color in block is set to By Layer, entire block color will follow layer color
- if geometry is set to By Block, then again block will follow layer color, but it can be overriden  in properties
- if part of geometry is set to By Layer and another part is set to By Block it will combine both behaviors

I will check later if it works the same in master.