LibreCAD/LibreCAD #1722: [Bug] Moving enities inside block makes copy of them

rola22, 2024-01-12:
Hello.
I found a bug while playing with blocks.
When i want to move enity inside block it gets copied instead of moved.

Steps to reproduce:
- create enity (for example circle, center at 0,0; radius 50) and make block of it,
- edit recently created block, move original circle to 25,0 (delete original), save block
- now you have block with two circles, one at 0,0 and second at 25,0

Linux Mint
LibreCAD:
Version: 2.2.1_alpha-327-g8e102459
Compiler: GNU GCC 11.4.0
Compiled on: Jan 12 2024
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Linux Mint 21.1

lordofbikes, 2024-01-13:
It works as expected in 2.2.0.2 release.
In latest master the problem occur.
The problem exists only in the current document window.
Saving the drawing, close and reopen it fixes the issue.

dxli, 2024-01-14:
The issue is caused by pen manipulation of RS_Insert member entities, so deleted entities are drawn with the pen of normal entities.

I'm not sure whether this should be fixed at RS_EntityContainer, RS_Insert, RS_GraphicView, or RS_Painter level.