LibreCAD/LibreCAD #1723: How to install latest vrsion to Ubuntu?

MikePryadko, 2024-01-14:
Hi everyone!
What is "right way" to install **latest** version on `Ubuntu`?

Official Ubuntu repo have old version (`2.1.3`). There is "custom" PPA, but it's even more outdated (last modified 5 year ago)

[image]

[image]

I've downloaded `LibreCAD-2.2.0.2-x86_64.appimage` but it doesn't install anything, just launching... :confused: 

[image]

What's wrong with it??

rola22, 2024-01-15:
Hello.
Appimage is portable file format which mean you don't install app. just run it :)

fa201, 2024-01-15:
I share a bit more details:

1. Go to gihtub Release page to get the appropriate release : https://github.com/LibreCAD/LibreCAD/releases
2. Download the file *.AppImage
3. Launch with double click on AppImage file or open via console with `./xxxx.AppImage` (xxx being the proper version).

One thought: I always have 2 rolling releases. The one tagged with 'latest' (green label) which is production and normally with less bugs and the other is the latest 'Continous build' to use new features or corrected bugs.

MikePryadko, 2024-01-16:
Thanks for answers but I think we should have latest version in some repo, not only "portable" binary. :neutral_face: How can we do it? Maybe responsible for it person needs some help?

lordofbikes, 2024-01-16:
Each Linux distribution has package managers who are responsible for updating or backporting software. Usually they are not related with the software project.
This is the Linux concept of distributions and security, which is more or less restrictive over various distributions. There are Linux distributions which already use LibreCAD 2.2.0, but obviously not Ubuntu.
So if you want to see a more recent LibreCAD version in Ubuntu, you have to get in touch with Ubuntu maintainers.

Maintaining a PPA needs skills and reputation on Launchpad, which is very different from GitHub workflows. Thus it is much easier for us to maintain an AppImage as binary package for many Linux distros.

Long story short, the responsible person for updating LibreCAD in Ubuntu is at Ubuntu.

closed 2024-01-16 by lordofbikes

lordofbikes, 2024-01-18:
The intention of deb-packages is to bundle an application only and define external dependencies.
That's the reason, why deb-packages are built for a single Linux version. Also, they might not be compatible between different distributions of the same version, as they possibly use different library versions.
So there is not one deb-package we have to deploy, we would need to build multiple packages, for different Linux versions and distributions. And this is only for Debian derivatives, what shall we do with other package formats, e.g. RPM?

The AppImage format was created because of those facts. A single AppImage is usable on many Linux distributions and version. And unlike deb-packages it is not limited to Debian derivatives.