LibreCAD/LibreCAD #1724: Librecad chrashes

jazzific, 2024-01-15:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

dxli, 2024-01-16:
0dac0b0 fixed one issue of the hatching loop extraction algorithm.

The other crashes are due to invalid hatching loops for the current algorithm:

Currently, for a hatch:
1. the edges must form simple loops;
2. each loop contains edges and the edges should only be joined at start/end points;
3. each loop must be closed;
4. edges of loops cannot share any common point except at start/end points, i.e. no intersection/overlap;
5. two loops may touch each other (an allowed exception of rule 4.), but cannot cross each other or itself;
6. Finally, as usual, the hatch size cannot be close to tolerance requirement (10^-10 is too small, and 10^10 is too large).

jazzific, 2024-01-16:
Thanks for working on this issue!
I'm not sure I really understand what you say: Is there a problem coming from the way I use hatching?

The way I do it is:
- draw a rectangle
- create a polyline from the segments
- select the polyline and hatch.

I just discovered that the "closed" checkbox is not always checked, but this does not solve the problem.
The error doesn't always happen though, but I cannot find why.

It happens when I click "ok" on the hatching pop up or when I double-click on the hatching to edit, which is why I suppose it might not come from the way I proceed...
I sometime get a segmentation error in the same way.

I do find very frustrating that Librecad closes instead of just showing an error, would it be possible to change that?

dxli, 2024-01-16:
The invalid contour contains many overlapping lines. For example, a rectangle shape may actually be formed by 300 overlapping lines.
It's possible to support overlapping lines, but it's a future feature.
Yes, we can improve the error handling by avoiding crashes and showing more meaningful error messages.

A quick workaround is to clean up the contours and recreate hatches

dxli, 2024-04-02:
Workaround exists. Input considered invalid due to overlapping edges. 

Closing now.

closed 2024-04-02 by dxli