LibreCAD/LibreCAD #1725: Duplicated menu item for Full Screen view option

kromanenko, 2024-01-20:
### Expected behavior
It is expected to see a single non-ambiguous toggle for the full-screen viewing option. 

### Observed behavior
Instead, I can observe two different menu items in the same menu: 
«**Fullscreen**» and «**Enter Full Screen / Exit Full Screen**». 
Note that keyboard shortcuts are different too: 
«**Control + Command + F**» and «**Fn + F**» accordingly. 
Both menu items and keyboard shortcuts are functioning identically. 

[image]

### Steps to reproduce
* Launch the LibreCAD. 
* Click «View» on the menu bar.  
* Observe actual behavior.

### LibreCAD version info
Version: 2.2.0.2
Compiler: Clang 13.0.0
Compiled on: Jul 29 2023
Qt Version: 5.15.10
Boost Version: 1.82.0
System: macOS 12.7

### Operating System info
  System Version:	macOS 12.7.2 (21G1974)
  Kernel Version:	Darwin 21.6.0
  MacBook Pro (14-inch, 2021)

dxli, 2024-01-21:
is this a config issue? I only got one Full Screen entry, without the last one shown at the bottom in your screenshot.

lordofbikes, 2024-01-21:
The upper one is created by LibreCAD, the lower one is probably added by MacOS.
I found no "Enter Full Screen" string in LibreCAD code base.

On Linux I also only have the upper one.

kromanenko, 2024-01-21:
Yes, it seems that 2nd option comes from the macOS.

dxli, 2024-05-16:
Yes, LibreCAD only adds the menu item "Fullscreen" (no space). The "Enter Full Screen" is from the OS

closed 2024-05-16 by dxli