LibreCAD/LibreCAD #1727: Lack of units in Image Export Options dialog

kromanenko, 2024-01-20:
### Expected behavior
Image export options should have units clearly specified. 

### Observed behavior
* Units of Bitmap Size "**Resolution:**" dialog are not clear. Implying dots per inch?
* Units of Bitmap Size "**Width:**" and "**Height:**" fields not specified. Implicitly, pixels, most likely. 
* Units of Borders "**Left / Right - Border:**" and "**Top / Bottom - Border:**" are not specified. Are they pixels or percents?
[image]

### Steps to reproduce
* Launch the LibreCAD with any drawing. 
* Navigate through menus: File -> Export ->  Export as Image.
* Click "Save".
* Observe the "Image Export Options" dialog. 

### LibreCAD version info
Version: 2.2.0.2
Compiler: Clang 13.0.0
Compiled on: Jul 29 2023
Qt Version: 5.15.10
Boost Version: 1.82.0
System: macOS 12.7

### Operating System info
System Version: macOS 12.7.2 (21G1974)
Kernel Version: Darwin 21.6.0
MacBook Pro (14-inch, 2021)

dongxuli-concord-design, 2024-03-03:
This is a feature request.

dxli, 2024-05-16:
@sand1024 , 

Would you like to get this one?

sand1024, 2024-05-17:
yes, I can take a look - yet not right now, in 1-2 days.