LibreCAD/LibreCAD #1728: Excessive status bar height

kromanenko, 2024-01-21:
The status bar height looks somewhat excessive, taking up quite a lot of space on the laptop screen. 
I remember this is a long-standing feature of LibreCAD UI rather than a bug. However, is it possible to make the status bar less beefy? 
For instance, here I highlighted the vertical gaps that could have been diminished. 
[image]

Also, it looks to me that mouse button action hints appear too high, above the upper string of the status bar. 
It is expected to see it in one level with the line of Cartesian coordinates. 
[image]

It seems that issue was partially addressed in https://github.com/LibreCAD/LibreCAD/pull/1410

### LibreCAD version info
Version: 2.2.0.2
Compiler: Clang 13.0.0
Compiled on: Jul 29 2023
Qt Version: 5.15.10
Boost Version: 1.82.0
System: macOS 12.7

### Operating System info
System Version: macOS "Monterey" 12.7.2 (21G1974)
Kernel Version: Darwin 21.6.0
MacBook Pro (14-inch, 2021)

lordofbikes, 2024-01-21:
This is a reasonable compromise for platform independence too.
In #1304 you can see what happens, when the lines are squished too much.
This is related with screen resolution (screen DPI) and OS text settings (font DPI).
Qt also has additional rules concerning font size and margins.

On other OS or different OS font settings the gaps are smaller. The minimum height was chosen with a safety offset to avoid clipping as reported in #1010.

Concerning the alignment between coordinates and mouse button hints, there is no relation between those widgets. There are always two lines in coordinates, but there can be 1-3 lines in mouse button hints.
From programming point of view it would need immense efforts to align those widgets.
The coordinates are composed with two single line text labels, arranged with Qt layout.
And the mouse button hint is a multi line label. Even when the mouse button hint has two lines, it would not be aligned with the coordinates because of invisible borders and margins from Qt layout in the coordinate widget.

kromanenko, 2024-01-21:
That's a pity we can't have more subtle status bar. However we can easily Disable/Enable the Status Bar at any moment.

dxli, 2024-01-28:
I found the mouse widget is spanning the whole status bar height.
We can:
1, reduce the mouse icon height in the status bar;
2, allow a vertical status bar;
3, redesign to use a single line status bar

kromanenko, 2024-01-28:
> I found the mouse widget is spanning the whole status bar height. We can: 1, reduce the mouse icon height in the status bar; 2, allow a vertical status bar; 3, redesign to use a single line status bar

I think that slightly reducing the mouse icon would be the simplest and the most effective way.

dxli, 2024-01-29:
need to test the impact on high dpi monitors

lordofbikes, 2024-01-29:
More important, with different font scale settings. Otherwise we switch back to #1010.
Also important to check standard status bar height, e.g. during auto save, or we get back #1026;
Possibly a 14" monitor is not representative to evaluate this.