LibreCAD/LibreCAD #1729: line is not detected whe using parallell

kmpm, 2024-01-23:
### Expected behavior
Using the parallel tool I should be able to hoover above or in the vicinity of a line that I want to make a parallel to and the suggested line should show.

### Observed behavior
Nothing happens in librecad but works like a charm in qcad 3.27.6

### Steps to reproduce or sample file
See attached sample. I can not create parallels to the lines in layer v0. No suggestion turns up. It's really hard to select them as entities as well.

### Operating System and LibreCAD version info
Windows 11 Pro 22H2

Version: 2.2.0.2-5-g3aa2c41c
Compiler: GNU GCC 13.2.0
Compiled on: Oct 11 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)

[sample.zip](https://github.com/LibreCAD/LibreCAD/files/14028682/sample.zip)

lordofbikes, 2024-01-23:
The problem is not the parallel tool, but the underlying image.
Duplicate of #1259

kmpm, 2024-01-24:
Tried the same _without_ an underlying image with same bad result

lordofbikes, 2024-01-24:
Worked for me as expected with 2.2.0.2-5-g3aa2c41c.
Used your sample with invisible image layer.
Maybe a too big/small parallel distance?