LibreCAD/LibreCAD #1731: Any interest in hierarchical Layers support in LibreCAD? Like this

sand1024, 2024-01-29:
Actually, this is not exactly the issue - currently LibreCAD does not support layers hierarchy and this is not too convenient (yes, of couse Autocad DXF does not support it too). 

However, it is possibel to emulate hierarchy by appropriate naming conventions for layers. 

So basically I've spent some time and created a widget (similar to Layer List) that does this and also supports a couple of nice operations (like move selection to layer, copy/duplicate layers and so on). 

I've uploaded a demo video with screencast that shows the widget on YouTube - https://www.youtube.com/watch?v=kFLRBU7FSCA. 
The list of implemented features - in video's description.  If you'll see some room for improvements - please let me know. 

In general, if such widget is usefull for somebody except me - I could commit into my repo and create merge request ther. If not - well, it will be just my modified copy of LibreCAD. 

P.S. not sure whetether the issue there is the right place for discussion of such things - if there is some more suitable place, please let me know. 

Thank you!

dxli, 2024-01-30:
This one looks great to me.

sand1024, 2024-01-30:
Appropriate code is in this repo https://github.com/sand1024/LibreCAD. 

All functionality was commited by one large commit - https://github.com/sand1024/LibreCAD/compare/80ed4c2fdecb86f4c3e3147ea65a92efa8cf0c98...08dac86914cfa95daea01843ce91b031e5582480

I'm not sure whether it's practical to create pull request at the moment, however, it will be great if somebody may check the implementation and let me know any feedback (both features and implementations). 

I think feedback with issues/discussions in that fork's repo will work better rather than directly there? Anyway, I'm more than open for the feedback and suggestions.

lordofbikes, 2024-01-31:
Many thanks for your intention to share this feature!

I've cherry picked your commit 80ed4c2fdecb86f4c3e3147ea65a92efa8cf0c98, the fix for `rs_actiondefault.h`, as this is another issue.
As old as QCAD and unbelievable, that this wasn't found yet :rofl:. Nice catch.

I haven't gone through the implementation yet, but I have two recommendations so far.
* To keep your `master` in sync with the main repo, you should move the feature commit into a separate branch. This makes pull request and merging much easier. So you should pull the new `master` with your fix and on this commit (which is different from yours, SHA de9cb37f0c1f36f683bf6d112227e9a5968677b5) create a branch for the feature.
Don't hesitate to ask if you need help with `git` here.
* For all the new files please replace the prefix `qg_` by `lc_`. `qg_` files are QCAD origins and for newly implemented files we use `lc_` to keep them apart.

sand1024, 2024-01-31:
Thank you for your comments. Yes, basically that issue with define was a bit annoying - I've commiteed initilal version of CMakeLists.txt also (yet without plugins build so far) - hope it may be good start for adding support of cmake for the project too. 

BTW, when I've tried to achive build of the project via cmake - I've noticed lots of files that are part of repo but are excluded from build. 

It seems that they are not needed so far, so probably it's better to move them in to some separate directory (if they are needed, say, for some historical reasons)? Such files are commented in CMakeLists.txt in that repo. 

Regarding the prefix - yes, got your point, I'll refactor naming accordingly. Indeed, it was not clear for me why there are different prefixes for classes, so I've just used conventions from List widget. 

If there are naming/style etc. conventions that should be followed to keep the overall codebase of the project - please let me know that. 

Also, I think there might be another generic issue with my current implementation - tha generic decomposition and organizing of the code. I'll be glad to know your opinion there. 

Actually, I've tried to affect the existing codebase as little as possible, and that's a reason why, for example, dialogs are shown not via QG_DialogFactory or why some operations are implemented not as actions but directly as slotts.  

Of course, existing code may be reworked in such way and that may be more consistent approach - however, I'm rather concerned thare that it may introduce some regression issues and will require additional testing. 

And indeed, good point regarding git - I'll follow the way you've mentioned, thank you.

lordofbikes, 2024-01-31:
Thanks for the changes! I think this is a good base now to inspect code and work on your forks branch until we have it ready to merge.
With nearly 8000 lines added, it may take some time to get behind the code.
Don't worry about code style. Our huge mature code base has so many styles meanwhile that it's not worth the efforts to unify them.

Concerning the unused files I assume they're fragments from earlier features or test code and were forgotten to delete. We have to check them in history, if they were ever used at all. Probably they can be deleted, git remembers them anyway.
Maybe some are only used with specific preprocessor macros defined, that has to be checked too.
But this is additional work for another branch or the main repo itself, we should not mix up separate concerns.

The direct link to the branch: https://github.com/sand1024/LibreCAD/commits/layer_tree_view/

Didier-H, 2024-02-10:
Hi there. This is exactly what I was looking for. How can I get a version with this widget to try it ?

dongxuli-concord-design, 2024-02-10:
Try to build from source.

I feel we should provide a simple script to build from scratch using any GitHub branch.

closed 2024-05-16 by dxli