LibreCAD/LibreCAD #1732: Wayland does not support QWindow::requestActivate()

yechs, 2024-01-31:
The program opens on Wayland, but crashes immediately when selects File>Open.

A temporary workaround is to start under x11: `XDG_SESSION_TYPE=x11 librecad` with xwayland (which works fine).

### Expected behavior

A new window should open.

### Observed behavior

Program crashes.

The console output is pasted below.
```
$ QT_QPA_PLATFORM=wayland librecad
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
QG_LayerWidget::update: nullptr lastLayer
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
qt.qpa.wayland: Wayland does not support QWindow::requestActivate()
The Wayland connection experienced a fatal error: Protocol error
```

### Steps to reproduce or sample file

1. Run librecad under Wayland
2. Select File>Open
3. Observe program to crash

### Operating System and LibreCAD version info

Arch Linux (rolling) with following relevant packages:
- librecad 2.2.0.2-2
- Wayland 1.22.0-1
- Gnome-shell 1:45.3-1
- xorg-xwayland 23.2.4-2

yechs, 2024-01-31:
May or may not be related: #1594

yechs, 2024-01-31:
Updating the title since I realized that the error message is printed every time, before the program crashes.

lordofbikes, 2024-01-31:
What do you expect shall we do?
LibreCAD does not have any low level code which is Wayland specific. We use standard C++ libraries and Qt libraries which abstract low level interfaces from our code.
As you already mentioned it runs on X11 and same code runs on various Windows and MacOS versions too.
I'm also not sure if Qt is the right address for this issue.
I think Wayland or possibly Arch maintainers are more competent for this issue.
The error message from Wayland supports this assumption too.

closed 2024-02-01 by yechs