LibreCAD/LibreCAD #1733: Cannot load a basic DXF file

5axes, 2024-02-01:
Cannot load a "basic" DXF file .

OS Windows 10 
LibreCad : [LibreCAD-Installer-x64-2.2.1_alpha-332-g5b50aa69.exe](https://github.com/LibreCAD/LibreCAD/releases/download/latest/LibreCAD-Installer-x64-2.2.1_alpha-332-g5b50aa69.exe)

[image]

[image]

[Урал 44202.zip](https://github.com/LibreCAD/LibreCAD/files/14126194/44202.zip)

lordofbikes, 2024-02-01:
It opens for me on Linux, with 2.2.0 release and latest master.
Probably it's an OS issue with non ASCII characters in the filename.
Please try to rename the file to something without local character set or numbers only.

The drawing has some issues, missing fonts and unsupported text formatting, but basically the drawing should be usable.

5axes, 2024-02-01:
It's ok for the file name ... but even it's an unsupported text formatting other Viewer can read it.

LibreCad
[image]

eDrawings
[image]
 
[44202.zip](https://github.com/LibreCAD/LibreCAD/files/14131140/44202.zip)

lordofbikes, 2024-02-02:
That's what I meant, missing fonts. LibreCAD has it's own font format and does not support standard fonts like TTF.
And the front dimensions has some formatting which is also not supported in LibreCAD, that's why the dimension label is much longer than in your screenshot.
For both issues we already have tickets (#1105, #1286), but no solution so far.

Sadly, that's the best LibreCAD can do for this drawing.

closed 2024-02-02 by lordofbikes

dellus, 2024-02-02:
44202.dxf: I exploded the inserted title/frame block. Clicked with "Properties" on one of the text rectangles. An MText input box appeared.
In the text field the text was shown. Font field was empty, height = 0. After defining a font and changing height to a proper value, the text appeared in the drawing.

dongxuli-concord-design, 2024-02-02:
Should we provide a default font size? If the font size was found to be zero.

Default font should be unicode.

Intentionally don't 0 to hide is not useful

5axes, 2024-02-02:
This seems a good principle in the absence of values, or if a value is set to zero for no logical reason, why not use the default value, the default font, etc.?  If an application is to be appreciated by users, it must be robust in reading and adapt to the errors of others (within reason), and flawless in writing.