LibreCAD/LibreCAD #1735: Small yet nice widget to add - pen palette

sand1024, 2024-02-11:
I've created a small yet nice widget for a bit more convenient way for working with pens. 

It allows managing of persistent pens, select entities based by pens and apply pens to selection, current layer and so on.  

Demo video for the widget is there - https://youtu.be/YSU0K3-PfbI
Feature branch - https://github.com/sand1024/LibreCAD/tree/pens_palette_widget
Readme - https://github.com/sand1024/LibreCAD/blob/pens_palette_widget/PEN_PALETTE_README.md

If it is better to create a merge request rather than issue there - please let me know.

dongxuli-concord-design, 2024-03-03:
Can you send in a Pull Request?

sand1024, 2024-03-04:
here it is - https://github.com/LibreCAD/LibreCAD/pull/1745

dxli, 2024-05-16:
Merged

closed 2024-05-16 by dxli