LibreCAD/LibreCAD #1736: Layer with Custom color don't preselect "Custom Color" on Edit Layer Property

FHenry, 2024-02-11:
### Expected behavior
Create a layer
Set a color as "Custom" with #a39c67 (for exemple)
Click on OK
Click on Edit layer property, the  preselected color should be "Custom"

### Observed behavior
Create a layer
Set a color as "Custom" with #a39c67 (for exemple)
Click on OK
Click on Edit layer property, the defaulted preselectd color is LIght Gray instead of "Custom

### Steps to reproduce or sample file

https://github.com/LibreCAD/LibreCAD/assets/1050053/4221ae87-1004-4c44-a1c6-8c46369847b6

### Operating System and LibreCAD version info
Version: 2.2.0.2
Compiler: GNU GCC 9.4.0
Compiled on: Jul 29 2023
Qt Version: 5.12.9
Boost Version: 1.75.0
System: Ubuntu 22.04.3 LTS

AppImage version

sand1024, 2024-02-12:
yes, I can also confirm this issue. If custom color is set to color combobox, and if it is does not match one of last picked custom colors (amount of them is 3) - the combobox just displays wrong color. 

Actually, we've committed a fix for this issue recently - yet so far it's in separate branch, and I hope it will be merged one day into master and become part of the release. 

For cherry pick - this issue is address by the folowing (and couple of previous) commits: 
https://github.com/LibreCAD/LibreCAD/commit/370f457eb953b5347ea98b9615c38bff9c486b36

sand1024, 2024-10-26:
@FHenry - please check on latest alpha and if this issue is already addressed - please close it.

FHenry, 2024-11-05:
@sand1024 
I confirm fix in 2.2.2_alpha-latest

closed 2024-11-05 by FHenry