LibreCAD/LibreCAD #1739: LibreCAD HATCH misrenders in AutoDesk software

Rodrigodd, 2024-02-16:
### Expected behavior

The following hatched shape should render correctly in other software:

[image]

### Observed behavior

[Autodesk Viewer](https://viewer.autodesk.com/) and [Autodesk DWG True View](https://www.autodesk.com/products/dwg-trueview/overview), renders it incorrectly:

[image]

Converting it to GeoJSON, using the GDAL project, generates a invalid Polygon, see https://github.com/OSGeo/gdal/issues/9240.

The problem appears to be that the Hatch entity in the DXF files contains a boundary formed by disjoint shapes, which is invalid? I am not well versed in the DXF specification, so I hope someone more experience here can answer that. It will also help with the GDAL issue above.

### Steps to reproduce or sample file

Draw two disjoint rectangles. Choose Tool->Hatch. Select Both rectangles. Press Enter. Choose Solid Fill, and OK.

sample file: [test.zip](https://github.com/LibreCAD/LibreCAD/files/14314325/test.zip)

### Operating System and LibreCAD version info

```
Version: 2.2.0.2
Compiler: GNU GCC 13.1.0
Compiled on: Jul 29 2023
Qt Version: 5.12.11
Boost Version: 1.75.0
System: Windows 10 (10.0)
```

It is actually Windows 11, but the About box reported it as Windows 10.

dxli, 2024-02-19:
it's rendered correctly in the lastest development version.

Rodrigodd, 2024-02-19:
@dxli Yes, but not in the Autodesk software. My point is that the HATCH boundary geometry may be invalid, at least according to Autodesk, which, I believe, owns the DXF format.

dxli, 2024-02-20:
dxf format specifies:

1. number of loops;
2. edges of a loop;
3. edge type (lines here);
4. edge line positions.

Beyond that, we don't simulate Autodesk behaviors. In this particularly case, valid loops can be created from edges, so hatch is created this way.

lordofbikes, 2024-02-20:
The DXF reference for HATCH SOLID is not very specific about the composition of boundary path data. There is much room for interpretation.
Basically the same applies as in https://github.com/OSGeo/gdal/issues/9240, namely that the created geometry is invalid.

So what LibreCAD can do is to prohibit the use of non intersecting/including multi loop boundaries for hatch. But somebody might see this as feature and will be disappointed when we change it.

As we know the code how the hatch is created, we also have the working code to render this as expected. But we don't know the closed code from Autodesk, so it would require some valuable efforts to reverse-engineer the Autodesk code to find a way to create this special case of HATCH in a DXF.

The conclusion is, that you simply create two hatch, one for each square, and you  will have a DXF which is valid for OSGeo, Autodesk and LibreCAD.

dxli, 2024-05-18:
Our rendering is consistent with the dxf file format. To generate an unreasonable output for this particular test case is not useful.