LibreCAD/LibreCAD #1751: segmentation fault when drawing angle lines

hapees, 2024-03-14:
### Expected behaviour
An empty drawing.
Draw horiz. line, draw vert. line, draw line at an angle.
Left Mouse Click draws line.
Right Mouse Click - next task. 

### Observed behaviour
An empty drawing.
Lines - Horizontal - Left Mouse Click draws line.
Right Mouse Click causes segmentation fault. 
(regardless of lengtt, snap point)

### Steps to reproduce or sample file
See observed behaviour. 

### Operating System and LibreCAD version info
OS: Fedora 39 6.7.9-200.fc39.x86_64
LibreCAD:
Version: 2.2.1_alpha-350-g311a96ba
Compiler: GNU GCC 13.2.1
Compiled on: Mar 14 2024
Qt Version: 5.15.12
Boost Version: 1.81.0

dxli, 2024-03-15:
82ab47a

More issues like this are expected. Need to move ownership of option widgets to parent actions

hapees, 2024-03-16:
Hm.. 
is fixed.
Thanks a lot :-)

dxli, 2024-04-23:
The option widget owneship fix for print preview:

https://github.com/LibreCAD/LibreCAD/commit/299445856b62e0907bc2c6bca9c600e3e9f8b50a

The logic and lifetime control is cleaner when the action actually owns the option widget

closed 2024-05-18 by dxli