LibreCAD/LibreCAD #1752: Dock Widgets expanded with Pen Palette and Layer Tree

dellus, 2024-03-27:
LibreCAD 2.2.1 alpha 359  appimage on Puppylinux Fossapup

With the introduction of pen palette and layer tree the minimum width of the right side dock widgets has expanded from about 246 to 413 pixels on my 1200 x 1920 screen. If I switch off these two I can return to the smaller width.
This is a problem with smaller screens at the cost of drawing area. Maybe the developer had a modern large screen where this is not so much an issue.

dongxuli-concord-design, 2024-03-28:
I have set those two widgets to be hidden by default.

One thing might help is to use tabbed widgets.

GSStnb, 2024-04-01:
From a clean install, LS ( alpha 369 ) starts up like shown in the attached screenshot (docks using most of the display).  The Pen Wizard on top, Pen Pallette in the middle and Library Browser / Block List / Layer List / Layer Tree on the bottom.
[image]

The minumum width of the docks is determined by the Layer Tree, then the Pen Palette,  It appears that the width is determined by the row of buttons on the respective docks.  The Layer Tree has the most buttons, followed by the Pen Palette, etc.  The Block List can be narrowed the most (excluding the command line).

I use tabbed widgets and the dock will expand to the widest enabled widget, even if it is to the "back"
The Layer Tree is a good alternative to the Layer List, but I agree with Dellus - the minimum width for the two docks are too wide.  Can the single row of buttons be made into two rows?

dxli, 2024-05-18:
This is now fixed in master and 2.2.1 branches

closed 2024-05-18 by dxli