LibreCAD/LibreCAD #1753: FREE SNAP toll doesn't work when pressing Spacebar

GuirfaCarnevalli, 2024-03-28:
FREE SNAP toll doesn't work when pressing Spacebar

### Expected behavior
Enabling and disabling freesnap when press spacebar

### Observed behavior
Nothing happens

### Operating System and LibreCAD version info
Version: 2.2.1_alpha-360-ge0f8a4e0
Compiler: GNU GCC 13.2.0
Compiled on: Mar 28 2024
Qt Version: 5.15.2
Boost Version: 1.75.0
System: Windows 10 Version 2009

dxli, 2024-03-28:
I don't remember this feature.

I suppose it's a shortcut of the spacebar:

The shortcut should switch between the current snap mode and snap free.

Sounds like a new feature to me.

GSStnb, 2024-03-29:
It can be enabled via Options -> Application Preferences -> Defaults, Keyboard Settings: "Toggle free snap mode when SPACE BAR is pressed" (my version of LC is 2.2.1_alpha-241-g871768d7).

GuirfaCarnevalli, 2024-04-01:
It's enabled but it doesn't work as it should here

dongxuli-concord-design, 2024-04-01:
Did you restart the app after enabling it?

GuirfaCarnevalli, 2024-04-01:
Yes, and it still has no changes on the behavior
On version 2.2.0 it works normally

GSStnb, 2024-04-01:
Confirmed, also with Version: 2.2.1_alpha-369-g1d3b0f18.  With "Toggle free snap ..." enabled in Appl Prefs, pressing the [Space Bar] activates the command line.  And, it happens even if "Evaluate commands..." is disabled (both "Keyboard Settings" are disabled).

GSStnb, 2024-04-01:
Version: 2.2.1_alpha-371-gf277540d, drawing window is active, and a command is typed ("li") and then [Space bar] is pressed:
- With neither enabled (checked) -> "li" appears at command prompt - not 'evaluated', no change to free snap
- With "Evaluate commands" enable -> "li" appears at command prompt - not 'evaluated', no change to free snap
- With "Toggle Free snap..." enabled  -> "li" appears at command prompt - not 'evaluated', no change to free snap
- With both enabled *  -> "li" appears at command prompt - not 'evaluated', no change to free snap

NB:
- LC was restarted after each change to the Keyboard Settings
- Each test performed with free snap off, and the again with free snap on
- *: Shouldn't these two be mutually exclusive (e.g. radio button)?

dongxuli-concord-design, 2024-04-01:
I realized enabling space in command line to trigger snapping free is bad. One action should have only one meaning. Typing space in command line shouldn't mean evaluation or snapping free.

It should be instinctive:

Space bar triggers snapping free only if command line is NOT activated. We need to disable activating command line then by space in this case. 

Typing a space in command line should mean space, instead of evaluating. Since this is already a feature, I won't touch it.

GSStnb, 2024-04-01:
I ensured that the drawing window still had focus when pressing the space bar in all cases.  While the command, 'li' , did appear at the command line, the command line dock did not have focus.

dxli, 2024-04-01:
With 1160d67 , this feature is only triggered when the command line is not activated, and a command message is displayed when snapping is changed.

This way, this feature is independent from triggering evaluating by space, by difference in the command line focus.

dongxuli-concord-design, 2024-04-01:
Works as intended to me.

Snap free within the toolbar actually has the lowest priority:

Only when other snapping candidate available in range, free snapping is used; this is actually logical, otherwise, snapping free must be exclusive with other snapping modes, but the end user knows combination of modes can be used. 

The spacebar snapping free is disabling all other snapping modes other than snapping free, so it's completely free.

closed 2024-05-18 by dxli