LibreCAD/LibreCAD #1754: "Exclusive Snap Mode" does not have a shortcut kety combination

GSStnb, 2024-03-29:
### Expected behavior
It should be possible to enable or disable "Exclusive Snap Mode" via a shortcut (preferably a key combination, e.g [Alt]-[Space bar] to be able to toggle off / on quickly).

### Observed behavior
No shortcut is assigned to "Exclusive Snap Mode" 

### Steps to reproduce or sample file
Not applicable - feature request

### Operating System and LibreCAD version info
Version: 2.2.1_alpha-241-g871768d7
Compiler: GNU GCC 11.3.0
Compiled on: Jul 22 2023
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Debian GNU/Linux trixie/sid

dxli, 2024-03-30:
Add a shortcut Alt-x, because ALT-Space triggers my window manager.

GSStnb, 2024-03-30:
[Ctrl]-[Space] ?  (because it is hard to mix the space bar  :-)

dongxuli-concord-design, 2024-03-30:
Better to make it configurable.
Ctl-space is often the default shortcut to change keyboard input methods, so a potential trouble for non-english users.

GSStnb, 2024-03-30:
Understood.  Generally speaking, can a command be assigned a command alias using a meta keys (Ctrl, Alt, ...)?  For example, if "Exclusive Snap" was given the typical text command and a two  letter short-cut, e.g. 'sx' (in keeping with the other snap short-cuts), could it then be assigned a meta-key combination is the alias file?

dongxuli-concord-design, 2024-03-30:
It's actually within the reach of the current codebase. It's about coding time of 1 week, but potentially long QA time. The limit is as usual due to development time available.

GSStnb, 2024-03-30:
I reached the limits of my coding ability with the updated commands a while back , but QA I can help with...

dxli, 2024-03-31:
With the fix of issue #1751, we are pushing for another bug fix. Please Help test the latest master for any serious issues:

1. The highest priority, no user data losing crash;
2. No serious regression preventing to perform a particular method.

I'm also thinking to move to Qt6 after this bug fix, with a preview release to expose new features added. Particular, new geometry features like parabola support, shear support(WIP).

If you have some high priority issues in mind, we may get it done for the preview release as well.

GSStnb, 2024-03-31:
That was likely my greatest 'annoyance' - not have a key combination for ExclSnap.  I do have a couple of other things I've thought "it would be nice if..."  I'll open another 'issue' later in the week.

closed 2024-04-18 by dxli