LibreCAD/LibreCAD #1755: Print Preview Crashes LibreCAD

GSStnb, 2024-03-31:
### Expected behavior
Clicking the **Print Preview** icon, or selecting **File -> Print Preview**, switches to the print preview window for the current drawing.

### Observed behavior
LibreCAD crashes immediately without any error message

### Steps to reproduce or sample file
1. Open a new or existing file
2. Click the Print Preview icon, or select File -> Print Preview

### Operating System and LibreCAD version info
Version: 2.2.1_alpha-369-g1d3b0f18
Compiler: GNU GCC 11.4.0
Compiled on: Mar 30 2024
Qt Version: 5.15.3
Boost Version: 1.74.0
System: Debian GNU/Linux trixie/sid

dongxuli-concord-design, 2024-03-31:
Bad regression.

dellus, 2024-03-31:
It doesn't happen to me with alpha 369 Appimage on Puppylinux Fossapup.

dellus, 2024-03-31:
I should have been more detailed: it's on Fossapup bare, which is a very cut-down version and does not contain QT, maybe that's the key.

GSStnb, 2024-03-31:
More testing of alpha 369 Appimage on virtual machines...
  works on MX Linux (VM)
  works on a Debian GNU/Linux trixie/sid (VM)

Time to rebuild my desktop  |-p

GSStnb, 2024-03-31:
One more test: I wiped my LibreCAD configuration and started with a new config and it worked fine - Print Preview appeared as it should.

dellus, 2024-04-16:
From alpha 400 on Print Preview crashed LibreCAD for me too. In alpha 411 it didn't, Print Preview came up. But if I try to change the scale - again crash.
Clearing all settings in Application Preferences and running in a fresh Puppylinux live environment didn't help.

By the way, for not to open a new issue, -update current pen by active layer pen- also crashes LibreCAD.

dongxuli-concord-design, 2024-04-16:
Might be several distinct issues. 

Let's handle one by one. Any simple steps for me to reproduce the crash?

dellus, 2024-04-16:
Start 2.2.1-alpha 411 appimage.
Load an existing drawing or make a new one.
Open Print Preview. It starts as expected.
Try to set a different print scale. As soon as you select that scale LibreCAD exites.

The other issue, which doesn't belong into this thread:
In a new or existing drawing press the 'update current pen by active layer pen' button (double green triangle) sitting next to the standard pen selection tabs, introduced recently with the new pen palette dock widgets: - Exit.

dxli, 2024-04-17:
@dellus ,

Cannot reproduce the ‘'update current pen by active layer pen'  issue.

sand1024, 2024-04-17:
> The other issue, which doesn't belong into this thread:
> In a new or existing drawing press the 'update current pen by active layer pen' button (double green triangle) sitting next to the standard pen selection tabs, introduced recently with the new pen palette dock widgets: - Exit.

Do you experience such issue with visible or hidden PenPalette widget? 

Actually, that action relies on functionality provided by the widget, and as far as I can see by one of intermediate commit, if the widget is closed it may work incorrectly. 

I think in several days I'll commit a fix for this,

dongxuli-concord-design, 2024-04-17:
@sand1024 
  
Could review the widget layout also? 
Currently, they might be too wide for people with narrow screens (1920 or less).

I have moved the pen pallet and layer tree into tabs, but more work may be needed here. 

Regards!

dellus, 2024-04-17:
In the 2.2.1-alpha 412 appimage changing the scale in Print Preview doesn't crash LC anymore, but has no effect at all.

The other issue, 'update current pen by active layer pen': Pen Palette and Layer tree are indeed disabled. But they seem not to exist in this 412 appimage, I cannot find any tabs or settings to enable them.

dongxuli-concord-design, 2024-04-17:
Need a setting in

External toolbars:

UsePenPallet=1

perepujal, 2024-04-26:
It crashes for me in commit 10dd2daed996a9fc5e8e42e8d80dfb0c45effe76, start LibreCAD, no need to open an existing drawing, click the "Update Current Pen by Active Layer' Pen"

I've run LC on valgrind and attached the output in case it helps.
HTH
Pere
[Librecad-Valgrind.log](https://github.com/LibreCAD/LibreCAD/files/15135708/Librecad-Valgrind.log)

perepujal, 2024-04-26:
Just checking the validity of penPaletteWidget seems to do the job for me, no more crashes:

`pere@lleopard:~/LibreCAD$ git diff
diff --git a/librecad/src/ui/qg_actionhandler.cpp b/librecad/src/ui/qg_actionhandler.cpp
index 644829a9..0cd71c1e 100644
--- a/librecad/src/ui/qg_actionhandler.cpp
+++ b/librecad/src/ui/qg_actionhandler.cpp
@@ -2018,7 +2018,8 @@ void QG_ActionHandler::slotPenCopy(){
 
 void QG_ActionHandler::slotPenSyncFromLayer(){
     LC_PenPaletteWidget* penPaletteWidget  = QC_ApplicationWindow::getAppWindow()->getPenPaletteWidget();
-    penPaletteWidget->updatePenToolbarByActiveLayer();
+    if (penPaletteWidget != nullptr)
+      penPaletteWidget->updatePenToolbarByActiveLayer();
 }
 
 void QG_ActionHandler::set_view(RS_GraphicView* gview)
`

closed 2024-05-18 by dxli