LibreCAD/LibreCAD #1756: v2.2.0.x AppImage does not run on Ubuntu 22.04

gw8484, 2024-04-02:
### Expected behavior
Normal app startup

### Observed behavior
App launch fails to start up.

### Steps to reproduce or sample file
 > ./LibreCAD-2.2.0.2-x86_64.AppImage 
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_shall_not_be_integrated : sqfs_open_image error: /usr/local/src/LibreCAD-2.2.0.2-x86_64.AppImage
AppImageLauncher error: appimage_shall_not_be_integrated() failed (returned -1)
Squashfs image uses (null) compression, this version supports only xz, zlib.
ERROR: appimage_is_terminal_app : sqfs_open_image error: /usr/local/src/LibreCAD-2.2.0.2-x86_64.AppImage
AppImageLauncher error: appimage_is_terminal_app() failed (returned -1)
execv error: No such file or directory

### Operating System and LibreCAD version info
Linux coffee 6.5.0-21-generic #21~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri Feb  9 13:32:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

AzteCypher, 2024-04-18:
Experiencing the same issue.

Ubuntu 22.04.4 LTS, 64bit

dxli, 2024-04-19:
I downloaded 22.04 iso, and tried the 2.2.0.x appimage. 

It works properly without crashing at startup l

I am trying to get snapcraft up. Anyone want to help?

dxli, 2024-04-30:
The latest development version AppImage works properly in Kubuntu-22.04

dxli, 2024-05-18:
With release candidates available for 2.2.1, closing this one

closed 2024-05-18 by dxli