LibreCAD/LibreCAD #1757: LibreCAD-2.2.1_alpha-389-g91b83aa2-x86_64.AppImage  Seg Faults on Startup

GSStnb, 2024-04-07:
### Expected behavior
LC runs happily :-)

NB: I haven't tried the last couple of coninuous builds, but it was working fine two or three builds ago...

### Observed behavior
It does not.  :-(

### Steps to reproduce or sample file
Delete configuration directory "~/.config/LibreCAD/"
Start LC at command prompt $ /opt/librecad/LibreCAD-2.2.1a-x86_64.AppImage
   The "Welcome" dialogue appears
   LC crashes after clicking "Ok" with the following 
```
...
base: unicode
base: symbol
font: /tmp/.mount_LibreCjOaDHG/usr/share/librecad/fonts/symbol_misc1.lff:
base: symbol_misc1
font: /tmp/.mount_LibreCjOaDHG/usr/share/librecad/fonts/symbol_misc2.lff:
base: symbol_misc2
font: /tmp/.mount_LibreCjOaDHG/usr/share/librecad/fonts/symeteo.lff:
base: symeteo
font: /tmp/.mount_LibreCjOaDHG/usr/share/librecad/fonts/symusic.lff:
base: symusic
font: /tmp/.mount_LibreCjOaDHG/usr/share/librecad/fonts/unicode.lff:
base: unicode
/tmp/.mount_LibreCjOaDHG/usr/share/librecad/patterns
/tmp/.mount_LibreCjOaDHG/usr/share/librecad/qm
QLayout: Attempting to add QLayout "" to QG_LibraryWidget "Library", which already has a layout
/tmp/.mount_LibreCjOaDHG/usr/share/librecad/library
Segmentation fault
```

### Operating System and LibreCAD version info
Debian 13 (Trixie)
Latest continuous build; LibreCAD-2.2.1_alpha-389-g91b83aa2-x86_64.AppImage

GSStnb, 2024-04-08:
The samething happens with the latest build; LibreCAD-2.2.1_alpha-399-g2d40e66c-x86_64.AppImage.  I also tried this build on a fresh Debian 13 VM and had the same result.  :-(

dxli, 2024-04-09:
fixed by d294244

closed 2024-04-09 by dxli