LibreCAD/LibreCAD #1759: 'Properties' doesn't show 'By layer' for an entity if so assigned but the end result

dellus, 2024-04-17:
Expected behavior: 'Properties' for an entity drawn with 'by layer' should show 'By layer'.

Observed behavior: 'Properties' for an entity drawn with 'by layer' shows the attributes assigned to the layer.

LibreCAD 2.2.1 alpha 412 appimage or earlier (since introduction of Layer Tree and Pen Palette).
Puppylinux Fossapup64

sand1024, 2024-04-18:
mmm - not sure I understand. In which place so you observe such a behavior? In properties dialog? In active pen buttons? 
Please add more details.

dongxuli-concord-design, 2024-04-18:
By layer
By block

I do remember there's something logic cleanup needed.

sand1024, 2024-04-18:
mmm - I think I finally got it - if properties dialog is invoked for the entity, "By layer/By Block" label is not shown anymore, only actual color name. 
Yes, it seems it's my fault related to setup of properties dialogs, I'll fix it. Actually, the intent was to show not a black/white icon for the color in dialogs, but actual color (that same logic as in active pen toolbar).  
Anyway, I'll correct this with separate pull request.

dongxuli-concord-design, 2024-04-18:
We can keep the color icon and the text "by Layer" for example. Hope it works out for all languages

sand1024, 2024-04-18:
yep, that was exactly my original intent - yet there is a bug in implementation I didn't notice.

dxli, 2024-04-19:
I disabled the PenPallet by default.

You can remove the setting, or enable it by:

```
[CustomToolbars]
UsePenPallet=1
```

dxli, 2024-05-16:
Tools > Modify > Attributes, color is shown as "Unchanged";

Tools > Modify > Properties, color is shown as the current value.

Two behavior is good (for different usages). Closing this one.

closed 2024-05-16 by dxli