LibreCAD/LibreCAD #1760: Exporting SVG not displaying dimensions

ezequielleonzybert, 2024-04-19:
### Expected behavior

### Observed behavior

### Steps to reproduce or sample file

### Operating System and LibreCAD version info

dxli, 2024-04-20:
The SVG maker in LibreCAD was created for CNC purposes, instead of a generic SVG generator.

It's possible to create a generic SVG version with better support for dimensions and texts.

ezequielleonzybert, 2024-04-20:
> The SVG maker in LibreCAD was created for CNC purposes, instead of a generic SVG generator.
> 
> It's possible to create a generic SVG version with better support for dimensions and texts.

One could simply hide the layers that don't want to send to the CNC machine.

dongxuli-concord-design, 2024-04-20:
Yes. I would pay more attention to preserving CNC support while adding new features. That means more development time, which is the actual limit here.

ezequielleonzybert, 2024-04-20:
> Yes. I would pay more attention to preserving CNC support while adding new features. That means more development time, which is the actual limit here.

Do you mean that previous svg exporter had the functionality of exporting dimensions and then they exclude that function in order of preserving CNC support?

dxli, 2024-04-20:
No. 

I mean:

1, the current maker only considers entities needed for CNC;
2, SVG supports dimensions! 
3, simply adding dimensions to the current maker is not desirable, for adding unwanted entities;
4, the proper way is to add a control to the current maker, or even in the form of a different maker with dimension, text, etc, support.

dxli, 2024-04-20:
@ezequielleonzybert 

Do you want to work on this one? 

To expand SVG support is a great feature. Ideally, we can make svg a CAD format for LibreCAD.

ezequielleonzybert, 2024-04-20:
> @ezequielleonzybert
> 
> Do you want to work on this one?
> 
> To expand SVG support is a great feature. Ideally, we can make svg a CAD format for LibreCAD.

I'd like to contribute. But I can see the project is already moving to version 3. Will this version become obsolete then?

dongxuli-concord-design, 2024-04-20:
Anything v2 can be ported to v3. 

V3 is not actively being developed now.

I found v2 is still very useful, so I keep working on it.

We should also release another stable release for v2 soon.