LibreCAD/LibreCAD #1762: Usability issue? Active action is not finished (and not unchecked in UI) on new document creation

sand1024, 2024-04-23:
Not sure whether this is intentional behavior, but it seems to be suspiciously similar to a bug.  Or at least, this is quite confusing behavior.

Start some action. While action is active, create new file or open existing one (without cancelling the action). 

### Expected behavior

Action is finished and is unchecked in UI. 

### Observed behavior

Action is still checked, and is still active after new file creation - but it is active in previous document, so this is quite confusing why it's not possible to do something in newly created document. 

### Steps to reproduce or sample file

https://youtu.be/-g_bfTCSa0M

### Operating System and LibreCAD version info

Windows, 2.2.0, current dev snapshot from master

dxli, 2024-04-28:
could be the same root cause as #1764

sand1024, 2024-10-19:
fixed with one of PR's

closed 2024-10-19 by sand1024